green-green-avk/AnotherTerm #1: On terminal it cannot access sdcard

GitHubOpenIssue, 2020-05-31:
:...green_green_avk.anotherterm $ cd /sdcard
:/sdcard $ ls
ls: .: Permission denied

green-green-avk, 2020-06-01:
1) See https://github.com/green-green-avk/AnotherTermLite/issues/1
2) Say me if you still need it.

GitHubOpenIssue, 2020-06-02:
When you said "I did not add READ_EXTERNAL_STORAGE and WRITE_EXTERNAL_STORAGE permissions intentionally in order to prevent a reasonable question: "Why a terminal application needs to access my storage?". Technically, they are dangerous. That's why I added $TERMSH pick/send/view/edit commands."

Users have access sdcard so it could be also dangerous if users don't know what they are doing and that happens in Android (it is not neccesary on any application) then excuse me but I don't find your argument very convining.

The advantage of accessing there are many, for example from distros access to downloaded files or other vias like email etc been used to programming or whatever you want to do with them within distros.

and as a middle ground like other applications in Android do and is to have access to a folder created for this purpose by the applications

Well what you decide but for me it is a serious issue

Thank you very much!

green-green-avk, 2020-06-03:
So, let ~eat bee~ it be:
* https://github.com/green-green-avk/AnotherTerm/commit/3be69f5be38f9b183c71e529cc57cade612b1238
* https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIc_release

Posted on Google Play, some patience waiting.

closed 2020-06-03 by green-green-avk