green-green-avk/AnotherTerm #4: Any plan to implement termux boot like plugin and features.

kcubeterm, 2020-07-26:
Termux is nice but that's not support in android 4, I use terminal emulator in my old phone and that application doesn't support this features, I doesn't carry them I just sshing into them with termux but in case if it discharged I have to setup it again all the procedure.
I host a simple golang web file manager in that .

Thanks :)

green-green-avk, 2020-07-26:
Yes. Maybe even next week.

green-green-avk, 2020-08-04:
Let's try <https://github.com/green-green-avk/AnotherTerm/releases>: a session can be started by an intent now.

[Tasker](https://tasker.joaoapps.com/) (supports Android 4.0.X) / [Automate](https://llamalab.com/automate/) / other similar apps can be used to start on custom events.

1) See the manual inside **Another Term**.
2) It's very possible that enabling auto-start *(in Auto-start Manager)* will be required for:
   1) The automation app (**Tasker** etc.);
   2) **Another Term** *(as in some devices not only `android.intent.action.BOOT_COMPLETED` but also any intents sent by applications started at boot are controlled in the same way before user interaction with them occurs)*.

Does it solve your problem?

kcubeterm, 2020-08-05:
I didn't try it, can you explain more about it.
Do i need Tasker or automate also

green-green-avk, 2020-08-05:
Docs page is ready.
See <https://green-green-avk.github.io/AnotherTerm-docs/start-a-session-by-an-intent.html#main_content>.

seeya, 2022-11-16:
> Docs page is ready. See https://green-green-avk.github.io/AnotherTerm-docs/start-a-session-by-an-intent.html#main_content.

Hi I'm trying to auto start a session but don't really get how to indicate the specific session I need. 
For example, when I first create a terminal and name it "test", how can I open it on boot?

What I'm trying to achieve is run a script automatically on boot and read serial data.

green-green-avk, 2022-12-05:
Um... Sorry for the late response.

`green_green_avk.anotherterm.intent.extra.FAV_TOKEN` extra is a password and pointer to a favorite in the same time.
Favorite settings has the "token" field (2nd from the top). Tap "Generate" (stars) then copy / paste where you want.

StickyDigit, 2023-09-09:
Should this work  as per the documentation with the current .redist flavour from F-Droid (MkIIIv83)?

I'm mostly using LineageOS 17.1 and 18.1 on a variety of devices and have installed the Android 10 essentials plug-in. I see the Token field in the Session edit screen, and can generate it and copy/paste etc. (Might be worth a hint in the doc that simply clicking the generated string will copy it to the clipboard.)

I've been trying to get it to behave with Activity Manager and Automation (both from F-Droid) but can't get past errors about ControlService not exisiting. I've tried various combinations, but not managed better than launching the SessionsActivity.

I don't have/want google services on most of my devices, so the Automate example is not workable, and in fact fails silently on my only play store capable device even when I've pasted the proper FAV_TOKEN for the chosen session and changed the app name to green_green_avk.anotherterm.redist

Paste-able examples for "adb shell am" might be more readily adapted to other tools or launching specific sessions via shell scripts, and arguably be more FOSS friendly.

green-green-avk, 2023-09-12:
1.
> and have installed the Android 10 essentials plug-in

**LineageOS** provides you with root access. Isn't it? So you could just ease the Linux SE rules and forget about **PRoot** as a nightmare.

2.
> ControlService not exisiting

It looks like nonsense because it means: non of the Android applications will work on your devices as the *service* is the fundamental component type. Possibly, I need to know exact error messages in order to solve this conundrum...

3.
> fails silently on my only play store capable device

I'll try to prepare a really detailed step-by-step guide. Something looks missed badly. What Android version do you have on the mentioned device?

4.
> Paste-able examples for "adb shell am"

I'll try to find some time to play with **LineageOS** (I hope, an emulator will be enough)... `am` cannot be executed by regular Android applications on official Android. **LineageOS** should not prohibit it though if I'm right about point 1.

StickyDigit, 2023-09-12:
> > and have installed the Android 10 essentials plug-in
> **LineageOS** provides you with root access... forget about **PRoot** as a nightmare.

By default, LineageOS has no su capability, though it's possible to turn on ADB as root. I have installed Magisk on some of my devices, but not all. I don't have a problem with installing the A10 essentials plug-in and have had no trouble with proot. The containers of alpine/debian etc. that I've installed all seem to work fine.

> > ControlService not exisiting
> It looks like nonsense...

I've not had much need to deal with intents before, and then only to capture them and strip out URL's and such. I may have tried all bar the right combination over the course of a day. Will have another crack at it as time permits and send you more detailed info about one attempt which fails that way. I think I'm missing something which would be obvious to an android app dev, but invisible to this old linux-er.

> > fails silently on my only play store capable device 
> I'll try to prepare a really detailed step-by-step guide.

A step-by-step would greatly help. Hopefully in ways that can be adapted to FOSS automation packages.

> Something looks missed badly.

Yup. I repeatedly missed something.

>What Android version do you have on the mentioned device?

The ones I've done this with are Android 11 (LineageOS 18.1), One with Google apps. and the other without. 
I also have some LOS17.1 (android 10), a couple of LOS 14.1 (android 7), and various older devices with factory firmware below Android 6. If I can make this work, it'll make useful the old devices which cannot be rooted or flashed.

> > Paste-able examples for "adb shell am"
> I'll try to find some time to play with **LineageOS** (I hope, an emulator will be enough)... `am` cannot be executed by regular Android applications on official Android. **LineageOS** should not prohibit it though if I'm right about point 1.

Good-ish news there. I tried 'am' from a local terminal (AnotherTerm of course) on the device. It will not run as regular user/app, "failure calling service activity". Plays nice after 'su' of course :) though that's not an option on some devices. The good-ish bit is that I successfully ran "am monitor"  via adb without root, (whoami reporting user 'shell').

I've just managed to revive an old unflashable Xperia, runnning android 4.1.2. It too let me run 'am' via non-root adb.

StickyDigit, 2023-09-12:
Quick follow-up on that.

Correction: Of course it's not the F-Droid version of AnotherTerm, but the izzysoft one.

- Automation from F-Droid, and AnotherTerm install and on that old 4.1.2 Xperia.
- `am` runs on a local (non-root) shell on the old Xperia,
- I can run `am start -a android.intent.action.VIEW` and get prompted for app to handle it.