green-green-avk/AnotherTerm #6: only crashes when run on Android < 4.4

guillaume-d, 2020-07-30:
Android 4.1.2 only here (because of the hardware keyboard!).
The crash seems to caused by some use of java.lang.Objects (from green_green_avk.anotherterm.App.onCreate()), which is only available as API 19: https://developer.android.com/reference/java/util/Objects

I can provide more logs if needed, please just tell me which log level or regex filter I should use.

If you need a tester for 4.1, please do throw debug builds at me (even unsigned ones), I would be glad to help!

AnotherTerm seems to be the only replacement for GNURoot and KBox and  for devices too old to run Termux, thank you for that! :+1:

green-green-avk, 2020-07-31:
Here it is: <https://github.com/green-green-avk/AnotherTerm/releases/download/MkIIIq_release/app-debug.apk>

Just use `green_green_avk.anotherterm` and *Error* log level as a Logcat filter.

green-green-avk, 2020-07-31:
Fortunately, Android Studio API<19 images are now working and I can test it.

<https://github.com/green-green-avk/AnotherTerm/commit/9bd44416de9a6476a4515477da0d3604f4abfcf1>
Just try <https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIr_release>

guillaume-d, 2020-08-01:
Thanks for the quick response, here are the logs (I did not filter in any way as it did not get much longer that way and might give you some more context/insights):

```
07-31 22:45:21.992 D/KeyguardViewMediator(  727): setHidden false
07-31 22:45:22.062 I/ActivityThread( 5941): Pub green_green_avk.anotherterm.fileprovider: androidx.core.content.FileProvider
07-31 22:45:22.072 D/KeyguardViewMediator(  727): setHidden false
07-31 22:45:22.082 W/dalvikvm( 5941): VFY: unable to resolve virtual method 510: Landroid/content/Context;.getExternalMediaDirs ()[Ljava/io/File;
07-31 22:45:22.102 I/ActivityThread( 5941): Pub green_green_avk.anotherterm.linksprovider: green_green_avk.anotherterm.LinksProvider
07-31 22:45:22.102 I/ActivityThread( 5941): Pub green_green_avk.anotherterm.streamprovider: green_green_avk.anotherterm.StreamProvider
07-31 22:45:22.152 W/dalvikvm( 5941): VFY: unable to resolve static method 27051: Ljava/util/Objects;.toString (Ljava/lang/Object;Ljava/lang/String;)Ljava/lang/String;
07-31 22:45:22.152 W/dalvikvm( 5941): VFY: unable to resolve static method 27051: Ljava/util/Objects;.toString (Ljava/lang/Object;Ljava/lang/String;)Ljava/lang/String;
07-31 22:45:22.162 W/dalvikvm( 5941): VFY: unable to resolve static method 27051: Ljava/util/Objects;.toString (Ljava/lang/Object;Ljava/lang/String;)Ljava/lang/String;
07-31 22:45:22.172 W/dalvikvm( 5941): Exception Ljava/lang/NoClassDefFoundError; thrown while initializing Lgreen_green_avk/anotherterm/FontsManager;
07-31 22:45:22.172 W/dalvikvm( 5941): threadid=1: thread exiting with uncaught exception (group=0x40f22438)
07-31 22:45:22.192 D/KeyguardViewMediator(  727): setHidden false
07-31 22:45:22.192 E/AndroidRuntime( 5941): FATAL EXCEPTION: main
07-31 22:45:22.192 E/AndroidRuntime( 5941): java.lang.ExceptionInInitializerError
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at green_green_avk.anotherterm.App.onCreate(App.java:80)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1014)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4292)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.app.ActivityThread.access$1400(ActivityThread.java:143)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1301)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.os.Handler.dispatchMessage(Handler.java:99)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.os.Looper.loop(Looper.java:137)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at android.app.ActivityThread.main(ActivityThread.java:4962)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at java.lang.reflect.Method.invokeNative(Native Method)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at java.lang.reflect.Method.invoke(Method.java:511)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at dalvik.system.NativeStart.main(Native Method)
07-31 22:45:22.192 E/AndroidRuntime( 5941): Caused by: java.lang.NoClassDefFoundError: java.util.Objects
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at org.apache.commons.lang3.StringUtils.joinWith(StringUtils.java:4838)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	at green_green_avk.anotherterm.FontsManager.<clinit>(FontsManager.java:35)
07-31 22:45:22.192 E/AndroidRuntime( 5941): 	... 13 more
07-31 22:45:22.233 E/android.os.Debug(  727): !@Dumpstate > dumpstate -k -t -z -d -o /data/log/dumpstate_app_error
07-31 22:45:22.453 D/KeyguardViewMediator(  727): setHidden false
```

guillaume-d, 2020-08-01:
OK, I wasn't fast enough for you :-) and I somehow missed your last reply: release MkIIIr does indeed work for me (tested https://github.com/green-green-avk/AnotherTerm/releases/download/MkIIIr_release/app-release.apk).
Maybe you could link this bug from the MkIIIr release's notes though?

What I could additional try so far is very nicely done, congrats and thanks for the very timely fix!

closed 2020-08-01 by guillaume-d