green-green-avk/AnotherTerm #7: snippet for linuxcontainers.org aborts with: open() fails with errno=21

guillaume-d, 2020-08-06:
The snippet `( S=install-linuxcontainers.sh ; "$TERMSH" copy -f -fu "https://raw.githubusercontent.com/green-green-avk/AnotherTerm-scripts/master/$S" -tp . && chmod 755 $S && ./$S -a alpine 3.10 )` generated from https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-under-proot.html#main_content fails here on Android 4.1.2:

`.: open() fails with errno=21`. Note I only tested it with Alpine 3.10 and 3.12 though.

AFAICS no warning or error logs were emitted while the command ran with the _MkIIIs_ release build.

Workaround used:
1. first download the script manually (using a web browser, the device's `curl` or some other means)
2. then run it with `sh /`path`/`to`/install-linuxcontainers.sh -a `[...].

green-green-avk, 2020-08-07:
*Hmm... Unfortunately, I can't repeat it with the Android Studio API16 x86 image... (fortunately, <https://github.com/green-green-avk/AnotherTerm-scripts/issues/1> I can)*

1) What was your current directory at this moment?
2) Was `install-linuxcontainers.sh` actually downloaded by `$TERMSH copy`?

if not:

3) Let's try `( S=install-linuxcontainers.sh ; "$TERMSH" cat "https://raw.githubusercontent.com/green-green-avk/AnotherTerm-scripts/master/$S" > ./$S && chmod 755 $S && ./$S -a alpine 3.10 )`.

guillaume-d, 2020-08-08:
> _Hmm... Unfortunately, I can't repeat it with the Android Studio API16 x86 image... (fortunately, [green-green-avk/AnotherTerm-scripts#1](https://github.com/green-green-avk/AnotherTerm-scripts/issues/1) I can)_

Maybe see at the end of this comment first, your `cat` suggestion worked for me.

>     1. What was your current directory at this moment?

u0_a209@android:/data/data/green_green_avk.anotherterm $

>     2. Was `install-linuxcontainers.sh` actually downloaded by `$TERMSH copy`?

No.

>     3. Let's try `( S=install-linuxcontainers.sh ; "$TERMSH" cat "https://raw.githubusercontent.com/green-green-avk/AnotherTerm-scripts/master/$S" > ./$S && chmod 755 $S && ./$S -a alpine 3.10 )` just in case.

Nice, this works! Is my device having `cat` but no `cp` part of the problem?

green-green-avk, 2020-08-08:
Looks like this situation is similar in some way to this: <https://green-green-avk.github.io/AnotherTerm-docs/issues.html#termsh>

The error is from here:
<https://github.com/green-green-avk/AnotherTerm/blob/4dc37d351f9aba4fff7aee4aba4e1988d4c1a1c7/app/src/main/java/green_green_avk/anotherterm/TermSh.java#L559>
from here:
<https://github.com/green-green-avk/AnotherTerm/blob/4dc37d351f9aba4fff7aee4aba4e1988d4c1a1c7/app/src/main/jni/main.cpp#L329>
The vendor is a good Samaritan who added restrictions to the kernel `open()` call for the sake of the Greater Good...

Let's try the debug version of the APK.

green-green-avk, 2020-08-09:
Let's try also this variant with the direct `readlink()` call:
[app-release.zip](https://github.com/green-green-avk/AnotherTerm/files/5046719/app-release.zip)
*(it seems, this Dalvik runtime implementation has some vendor imposed modifications)*.

green-green-avk, 2020-08-10:
It would also be good to check this release candidate: [app-release.zip](https://github.com/green-green-avk/AnotherTerm/files/5048860/app-release.zip) with the final version of the fix.

green-green-avk, 2020-08-13:
Seems finally solved...
<https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv_release>

Please, confirm.

guillaume-d, 2020-08-14:
TL;DR: MkIIIv release version worked :+1:, only one _debug_ version worked (MkIIIu, older ones untested), all other (release) versions you mentioned since my last comment did not and logged nothing AFAICS.

On all my three successful tries the console output looked as follows (MkIIIu copies wrapped lines perfectly indeed! :+1:):

```
u0_a209@android:/data/data/green_green_avk.anotherterm $
t.com/green-green-avk/AnotherTerm-scripts/master/$S" -tp . && chmod 755 $S && ./$S -a alpine 3.10 )             <

Actual name: linuxcontainers-alpine-3.10

To uninstall: run `rm -rf "$DATA_DIR/proots/linuxcontainers-alpine-3.10"'

Creating favorites...
Done.

Arch: armv7a
Variant: -pre5
Root FS: alpine 3.10

write failed: EPIPE (Broken pipe)
Cannot download index from linuxcontainers.org
Source: https://us.images.linuxcontainers.org//images/alpine/3.10/armhf/default/20200811_13:25//rootfs.tar.xz

Getting minitar...
Getting PRoot...
Getting Linux root FS...
Setting up run script...
Configuring...

Done!

u0_a209@android:/data/data/green_green_avk.anotherterm $
```

Surprisingly, it seems to work fine afterwards despite the `EPIPE` error above.
(IIRC the same error message appeared with the `cat` workaround you suggested a few comments ago, I just forgot to mention it until now.)

Note: It seems impossible ATM to install both the debug package and the release package of the app at the same time: is this intentional? Given the substantial amount of private data the app may hold to (proots...), this may make testing debug builds impractical over time for most proot users. See https://developer.android.com/studio/build/application-id (look for `debug` there).

Thanks for all your tries, and sorry for the late feedback, I got caught up in other things.

closed 2020-08-14 by guillaume-d