green-green-avk/AnotherTerm #8: [FR] Use system (or hardware) keyboard first

sergey1369, 2020-08-13:
Please add an option to use system (hardware) keyboard first.

Someone may want use MessagEase by default, it has ctl/esc keys and is far more convenient 
for one hand input. Also there are bluetooth/deskdock keyboards, which do not use screen space.

green-green-avk, 2020-08-13:
How critical is the <kbd>Home</kbd> key support?
Do deskdocks of some Android devices has proper <kbd>Home</kbd> mapping that makes it visible for applications?
Is there some reason to implement own keyboard driver to make <kbd>Home</kbd> properly functioning for all Android devices?

sergey1369, 2020-08-13:
Oh, this fancy button, it is not really needed. There is linux shortcut CTRL-A for bash.

guillaume-d, 2020-08-14:
> Someone may want use MessagEase by default, it has ctl/esc keys and is far more convenient
> for one hand input. Also there are bluetooth/deskdock keyboards, which do not use screen space.

Hardware keyboard user here! I understand supporting these may be (very) low priority given the small user base and the difficulty to test without real hardware, but maybe the following could be done:
1. concerning this bug: detect external/hardware keyboard configuration changes (see https://developer.android.com/guide/topics/manifest/activity-element#config and for example https://developer.android.com/reference/android/content/res/Configuration#HARDKEYBOARDHIDDEN_NO) to make the virtual keyboard (dis)appear when appropriate
2. maybe related: ATM I have to change the default input method away from "Samsung keyboard" (to "Swype") in the Android settings for letters not to be input twice when typing on the hardware keyboard (only had this problem with AnotherTerm so far)

green-green-avk, 2020-08-15:
>     2. maybe related: ATM I have to change the default input method away from "Samsung keyboard" (to "Swype") in the Android settings for letters not to be input twice when typing on the hardware keyboard (only had this problem with AnotherTerm so far)

What Samsung exactly do you mention? I did some tests with a hardware keyboard but it's a new information for me.
Did you experience it with any other Terminal applications?

guillaume-d, 2020-08-16:
The hardware-keyboard-equipped device is a Samsung Galaxy Stratosphere II (AKA SCH-I415) with Android 4.1.2 (not rooted).

The problem only occurs with AnotherTerm AFAICS, at least he following apps are fine (I linked where I got them from):

- (jackpal's) Terminal Emulator 1.0.70: https://f-droid.org/en/packages/jackpal.androidterm
- ConnectBot 1.9.6-oss; https://f-droid.org/en/packages/org.connectbot/
- DroidVim 1.1.84 (outdated, I know): https://apkpure.com/droidvim/com.droidvim

green-green-avk, 2020-08-30:
Please, try <https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv5_release>

sergey1369, 2020-08-31:
Now works great! Thank you!

closed 2020-08-31 by green-green-avk