green-green-avk/AnotherTerm #10: Input chars duplication with some IMEs

green-green-avk, 2020-08-19:
The hardware-keyboard-equipped device is a Samsung Galaxy Stratosphere II (AKA SCH-I415) with Android 4.1.2 (not rooted).

The problem only occurs with AnotherTerm AFAICS, at least he following apps are fine (I linked where I got them from):

- (jackpal's) Terminal Emulator 1.0.70: https://f-droid.org/en/packages/jackpal.androidterm
- ConnectBot 1.9.6-oss; https://f-droid.org/en/packages/org.connectbot/
- DroidVim 1.1.84 (outdated, I know): https://apkpure.com/droidvim/com.droidvim

_Originally posted by @guillaume-d in https://github.com/green-green-avk/AnotherTerm/issues/8#issuecomment-674536098_

green-green-avk, 2020-08-19:
Thank you for the clarification.
Let's try <https://github.com/green-green-avk/AnotherTerm/releases/tag/%2310-1>

guillaume-d, 2020-08-21:
Unfortunately #10-1 behaves exactly the same as MkIIIv in all configurations I tested:

- `Samsung keyboard` IME (with visible or hidden virtual keyboard, with release or debug build): all letters typed are always input exactly twice, but more of them can be input without releasing the letter key (they "repeat") 
- other IME (`Swype`) (same variations): all letters typed are input once, but only as soon as they are released and therefore do not repeat

Note the physical keyboard's arrow keys and return key behave differently: in _both_ IMEs they always are emitted once correctly, but they can never repeat (same as `Swype` IME above).

HTH.

guillaume-d, 2020-08-23:
Thanks a lot, this #10 test 3 build fixed all problems in both IMEs (with hidden and closed virtual keyboard): char duplication is gone, key repeating now works! Well done! 👍

I opened #11 for the other Fn key problem.

green-green-avk, 2020-08-24:
Finally done: <https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv2_release>

closed 2020-08-25 by green-green-avk