green-green-avk/AnotherTerm #11: physical keyboard's Fn key ineffective

guillaume-d, 2020-08-23:
On my physical keyboard (QWERTY with Spanish letters) I cannot type any non-ASCII characters and most non-alphanumeric ASCII characters, because they are only accessible using a physical Fn key, which ATM does not work. This is broken only in AnotherTerm, the other apps listed in #10 work as expected.
BTW some Fn+letter combinations seems to have some effect:

-some "ring" the visual green bell at the right end of the line
- some delete the whole command line

I cannot find any pattern about which does what.

_Originally posted in https://github.com/green-green-avk/AnotherTerm/issues/10#issuecomment-678651498_

green-green-avk, 2020-08-23:
Please, use the **IME Test** app once more in the "cooked" mode. Only one <kbd>Fn</kbd>+key press is enough.

guillaume-d, 2020-08-23:
OK, Fn+s it is:

```

=====
Using raw mode
=====

=====
Using raw mode
=====

=====
Using cooked mode
=====
===
---
green_green_avk.imetest.InputView.access$000(InputView.java:20)
green_green_avk.imetest.InputView$1.finishComposingText(InputView.java:151)
android.view.inputmethod.InputMethodManager.checkFocusNoStartInput(InputMethodManager.java:1292)
android.view.inputmethod.InputMethodManager.checkFocus(InputMethodManager.java:1245)
android.view.inputmethod.InputMethodManager.access$000(InputMethodManager.java:203)
android.view.inputmethod.InputMethodManager$H.handleMessage(InputMethodManager.java:416)
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loop(Looper.java:137)
android.app.ActivityThread.main(ActivityThread.java:4962)
java.lang.reflect.Method.invokeNative(Native Method)
java.lang.reflect.Method.invoke(Method.java:511)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
dalvik.system.NativeStart.main(Native Method)
===
---
green_green_avk.imetest.InputView.access$000(InputView.java:20)
green_green_avk.imetest.InputView$1.beginBatchEdit(InputView.java:96)
android.view.inputmethod.BaseInputConnection.finishComposingText(BaseInputConnection.java:269)
green_green_avk.imetest.InputView$1.finishComposingText(InputView.java:152)
android.view.inputmethod.InputMethodManager.checkFocusNoStartInput(InputMethodManager.java:1292)
android.view.inputmethod.InputMethodManager.checkFocus(InputMethodManager.java:1245)
android.view.inputmethod.InputMethodManager.access$000(InputMethodManager.java:203)
android.view.inputmethod.InputMethodManager$H.handleMessage(InputMethodManager.java:416)
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loop(Looper.java:137)
android.app.ActivityThread.main(ActivityThread.java:4962)
java.lang.reflect.Method.invokeNative(Native Method)
java.lang.reflect.Method.invoke(Method.java:511)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
dalvik.system.NativeStart.main(Native Method)
===
---
green_green_avk.imetest.InputView.access$000(InputView.java:20)
green_green_avk.imetest.InputView$1.endBatchEdit(InputView.java:102)
android.view.inputmethod.BaseInputConnection.finishComposingText(BaseInputConnection.java:271)
green_green_avk.imetest.InputView$1.finishComposingText(InputView.java:152)
android.view.inputmethod.InputMethodManager.checkFocusNoStartInput(InputMethodManager.java:1292)
android.view.inputmethod.InputMethodManager.checkFocus(InputMethodManager.java:1245)
android.view.inputmethod.InputMethodManager.access$000(InputMethodManager.java:203)
android.view.inputmethod.InputMethodManager$H.handleMessage(InputMethodManager.java:416)
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loop(Looper.java:137)
android.app.ActivityThread.main(ActivityThread.java:4962)
java.lang.reflect.Method.invokeNative(Native Method)
java.lang.reflect.Method.invoke(Method.java:511)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
dalvik.system.NativeStart.main(Native Method)

=====
Using cooked mode
[... 341 lines cut ...]
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loop(Looper.java:137)
android.app.ActivityThread.main(ActivityThread.java:4962)
java.lang.reflect.Method.invokeNative(Native Method)
java.lang.reflect.Method.invoke(Method.java:511)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
dalvik.system.NativeStart.main(Native Method)
===
---
green_green_avk.imetest.InputView.access$000(InputView.java:20)
green_green_avk.imetest.InputView$1.endBatchEdit(InputView.java:102)
android.view.inputmethod.BaseInputConnection.finishComposingText(BaseInputConnection.java:271)
green_green_avk.imetest.InputView$1.finishComposingText(InputView.java:152)
android.view.inputmethod.InputMethodManager.reportFinishInputConnection(InputMethodManager.java:749)
android.view.ViewRootImpl$ViewRootHandler.handleMessage(ViewRootImpl.java:3178)
android.os.Handler.dispatchMessage(Handler.java:99)
android.os.Looper.loop(Looper.java:137)
android.app.ActivityThread.main(ActivityThread.java:4962)
java.lang.reflect.Method.invokeNative(Native Method)
java.lang.reflect.Method.invoke(Method.java:511)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1004)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:771)
dalvik.system.NativeStart.main(Native Method)
```

green-green-avk, 2020-08-23:
Hmm...

Either do we need to have an option for the <kbd>Alt</kbd> behavior?
Or just treat any left <kbd>Alt</kbd> in a special way if the `deviceId==0` (<https://developer.android.com/reference/android/view/KeyCharacterMap#BUILT_IN_KEYBOARD>)?
Ambiguous...

Surprisingly, the IME dispatches the event directly to the view around its `InputConnection` interface, so it's supposed to be done by `event.getUnicodeChar()` thus using the keyboard mapping table.

So at the moment your <kbd>Fn</kbd> acts exactly as the <kbd>Alt</kbd> supposed to do (i.e. prepending any ASCII symbol by ESC with one detail):
<https://github.com/green-green-avk/AnotherTerm/blob/7a656fbfc62daa693aaf77ee897df35512345d50/app/src/main/java/green_green_avk/anotherterm/ConsoleOutput.java#L102>
(as for any usual external keyboard the <kbd>Shift</kbd> must be the only modifier to be taken into account here).

Raw mode:
```
KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_G, scanCode=34, metaState=META_ALT_ON|META_ALT_LEFT_ON, flags=0x8, repeatCount=0, eventTime=6591161, downTime=6591161, deviceId=0, source=0x301 }
---
green_green_avk.imetest.InputView.onKeyPreIme(InputView.java:81)

...

KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_G, scanCode=34, metaState=META_ALT_ON|META_ALT_LEFT_ON, flags=0x8, repeatCount=0, eventTime=6591161, downTime=6591161, deviceId=0, source=0x301 }
---
green_green_avk.imetest.InputView.onKeyDown(InputView.java:251)
android.view.KeyEvent.dispatch(KeyEvent.java:2715)
android.view.View.dispatchKeyEvent(View.java:7262)
...
android.os.Handler.dispatchMessage(Handler.java:99)
```

Cooked mode:
```
KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_S, scanCode=31, metaState=META_ALT_ON|META_ALT_LEFT_ON, flags=0x8, repeatCount=0, eventTime=21063090, downTime=21063090, deviceId=0, source=0x301 }
---
green_green_avk.imetest.InputView.onKeyPreIme(InputView.java:81)

...

KeyEvent { action=ACTION_DOWN, keyCode=KEYCODE_S, scanCode=31, metaState=META_ALT_ON|META_ALT_LEFT_ON, flags=0x8, repeatCount=0, eventTime=21063090, downTime=21063090, deviceId=0, source=0x301 }
---
green_green_avk.imetest.InputView.onKeyDown(InputView.java:251)
android.view.KeyEvent.dispatch(KeyEvent.java:2715)
android.view.View.dispatchKeyEvent(View.java:7262)
...
android.os.Handler.dispatchMessage(Handler.java:99)
```

green-green-avk, 2020-08-23:
May be we need also to think about possible remapping <kbd>Alt</kbd> and mapping <kbd>Ctrl</kbd> and <kbd>Insert</kbd> somewhere?

green-green-avk, 2020-08-24:
As a concept: <https://github.com/green-green-avk/AnotherTerm/releases/tag/%2311-1>
Look into settings.

guillaume-d, 2020-08-24:
#11 test 1 fully worked once I activated the new setting! 👍
Note the checkbox labels are pretty long and look truncated: they end with "acts as.." and "and similar" here.
One more time thanks a lot!

> May be we need also to think about possible remapping Alt and mapping Ctrl and Insert somewhere?

I do not mind about losing Insert or even Alt, but having Ctrl is crucial, and (at least in Vi) it can also compensate for other missing keys (Esc: Ctrl-C, Tab: Ctrl-I).
Some way of mapping physical keys or buttons to PC keys would be perfect, even when only over a textual configuration file. The Key Mapping Editor might already support this, but I am currently unsure how to use it, more documentation about it could help.
One should be able to map at least Volume-Up, Volume-Down and Back (should be present on many devices), ideally also the hardware keyboards' exotic keys ("Search", "Chat", `web` (sic!) and `OK` keys here) by entering their key code.
I won't push it :wink: by asking for the Menu button which is probably almost nowhere else to be found nowadays, or for the Power and Home ones which IIRC need convoluted tricks...

green-green-avk, 2020-08-24:
> ...ideally also the hardware keyboards' exotic keys...

Logging them in the "raw" mode could be a good idea. Just to estimate the possibility at least.

guillaume-d, 2020-08-25:
> > ...ideally also the hardware keyboards' exotic keys...
> 
> Logging them in the "raw" mode could be a good idea. Just to estimate the possibility at least.

Here you are then:
- "Search" is KEYCODE_SEARCH
- "Chat" outputs nothing at all in the IME Test app, but directly opens the stock Messaging app, so that one key is probably hopeless
- `web`: same as above (opens the stock web browser app)
- `OK` is KEYCODE_DPAD_CENTER
- ...and Menu is KEYCODE_MENU as one would expect

(Please tell me if you'd need more logs from the IME Test app.)

However listing each and every physical key Android ever defined for mapping in the UI might be too much given the few users concerned.
(Note that the other apps listed in https://github.com/green-green-avk/AnotherTerm/issues/10 all do something similar but all leave out some keys that nobody claimed, and their UI is already pretty crowded.)

As said in https://github.com/green-green-avk/AnotherTerm/issues/11#issuecomment-679350501 a more generic solution might be more appropriate: I for one would not mind having to install IME Test and copy part of the results from there during the inital configuration to get my own keys working.

closed 2020-08-28 by green-green-avk

guillaume-d, 2020-08-28:
> #11 test 1 fully worked once I activated the new setting! 👍
> Note the checkbox labels are pretty long and look truncated: they end with "acts as.." and "and similar" here.
> One more time thanks a lot!

The original issue is also fixed in the official MkIIIv4 release, and the labels are not truncated anymore => closing.