green-green-avk/AnotherTerm #12: Mapping of built-in keyboard keys as modifiers

green-green-avk, 2020-08-28:
...
> May be we need also to think about possible remapping Alt and mapping Ctrl and Insert somewhere?

I do not mind about losing Insert or even Alt, but having Ctrl is crucial, and (at least in Vi) it can also compensate for other missing keys (Esc: Ctrl-C, Tab: Ctrl-I).
Some way of mapping physical keys or buttons to PC keys would be perfect, even when only over a textual configuration file. The Key Mapping Editor might already support this, but I am currently unsure how to use it, more documentation about it could help.
One should be able to map at least Volume-Up, Volume-Down and Back (should be present on many devices), ideally also the hardware keyboards' exotic keys ("Search", "Chat", `web` (sic!) and `OK` keys here) by entering their key code.
I won't push it :wink: by asking for the Menu button which is probably almost nowhere else to be found nowadays, or for the Power and Home ones which IIRC need convoluted tricks...

_Originally posted by @guillaume-d in https://github.com/green-green-avk/AnotherTerm/issues/11#issuecomment-679350501_

guillaume-d, 2020-09-08:
Would you accept a PR for this? (I do not have anything ready yet, I have just looked at the code and I have at least some idea how to proceed.)

I doubt I would have enough time (and expertise ATM) to do the UI for it, but I could at least contribute support for a separate configuration file for the key mappings which could be manually edited or copied to the right place inside the app.
That way also the additional maintenance burden for built-in hardware keyboard support would be minimal.

green-green-avk, 2020-09-09:
Hmm... A day or two and I could provide some preliminary solution to extend with a PR.

green-green-avk, 2020-09-12:
Please, test: <https://github.com/green-green-avk/AnotherTerm/releases/tag/%2312-1>
The UI is in global settings.
If you want to pass a key around IME (<kbd>Ctrl</kbd> can be sensitive to it in some old devices) just map it to itself (there was a hardcoded rule for <kbd>Ctrl</kbd> before).
* "Turn off" means "Completely mute a key";
* "Don't use as a terminal modifier" means "Bypass as is and use to determine the final key code only" (this option is for modifiers only).
In your case you need to map <kbd>Fn</kbd> (Left Alt) as "Don't use as a terminal modifier".
All other keys - as you wish. Yes, technically you can map even your <kbd>Back</kbd> button.

green-green-avk, 2020-09-12:
And yes, I'll finally fix the settings UI drop down lists rendering just a bit later.

green-green-avk, 2020-09-12:
Fixed: <https://github.com/green-green-avk/AnotherTerm/releases/tag/%2312-2>

green-green-avk, 2020-09-14:
<https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv9_release>

guillaume-d, 2020-09-15:
TL;DR: I cannot test MkIIIv9 because the `Hardware keys mapping` view's text takes too much screen space. 

If I force the orientation to be portrait(\*), I can see the `+` button, add a mapping and just barely see the added mapping line. Modifying the line afterwards is practically impossible.
(\*): Sliding the keyboard open forces landscape orientation here, but I'd like to map keys on the keyboard too, so I had to force portrait using a rotation manager app.

With landscape orientation, the `+` button is not even visible.

Maybe the help text could be made accessible over a `?` button like in the `Key Mapping Editor` or `Scratchpad` views?

Anyway thanks a lot again for working on that, I really appreciate it! :+1:

green-green-avk, 2020-09-16:
<https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv10_release>
Corrected.

guillaume-d, 2020-09-17:
In MkIIIv10 the `Hardware keys mapping` view's layout got a bit more usable, but mapping to modifier keys should IMHO be tweaked:

The layout still has a few problems (both landscape and portrait orientations):

- long "mapping target" labels (`MOVE HOME`, `MOVE END`, `PAGE UP`, `PAGE DOWN`) are initially truncated in the popup menu
- once one of these long labels are selected, the screen space taken never shrinks even after selecting a shorter label, thereby making the trash can icon invisible => the mapping cannot be deleted anymore
- also the sorting order used when mappings are added is not user-friendly (keycode-sorted?): maybe sort by the mapped key's label?

Ideas:

- maybe just use shorter labels: `HOME`, `END` and to a lesser extent `PG UP`, `PG DN` are pretty self-explanatory by (English (laptop)) keyboard standards.
- the right arrow glyph on each mapping line could be smaller also in these cases (the unconstrained size is pretty big)
- in landscape mode the margins/paddings between the settings tabs on the left and the actual mapping UI could be narrower

Apart from those issues, the layout stays usable even when defining a lot a mappings: there are still a few (scrollable) lines of help text, and the mappings list is scrollable too, well done! :+1:

All in all I could map all special keys of the built-in keyboard and almost all Android standard keys (including `Menu`, but neither `Home` nor `Power` as expected) that AnotherTerm labels as `External`: hooray! :+1:

However IMHO keys that are mapped to modifier keys like `Ctrl` should not need to be held while hitting the modified key: it makes typing rather hard and I think no other terminal app behaves like this.

green-green-avk, 2020-09-18:
Hmm... What's better: either release a toggled modifier on a first regular key press or not?

BTW, there is no reliable way to distinguish a built-in and an externally connected devices until API 29.

guillaume-d, 2020-09-19:
Maybe a double press on a modifier key should hold it, and a single press afterwards should release it (like Android does for `Shift`, and like jackpal's Terminal Emulator does for `Shift` *and* `Ctrl`)?

If that is too much to implement, maybe rather release the modifier after the 1st regular key, but fans of Emacs keybindings may differ... Also for `Shift` it could probably be the opposite, like some other lesser Android terminal apps do IIRC.

On detecting built-in keyboards: maybe yet another preference flag is needed then... :-/

green-green-avk, 2020-09-19:
Possibly intermittent variant as settings UI has definitely required to be fixed:
<https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv13_release>
"🔒<sup>1</sup>" - unsticks on first non-modifier key or doesn't stick at all if the key is pressed when the modifier is still being held.
"🔒" - just toggle.

guillaume-d, 2020-09-20:
Tested MkIIIv14 (after MkIIIv13), the settings UI is almost perfect now :+1:, and I am happy already only with "Ctrl🔒¹".

However:
- [minor nit:] in the settings UI I still find the key mapping sort order confusing
- the "lock indicator overlay" should IMHO be much less distracting, a smallish icon to the top right like when hitting `Ctrl+I` is maybe enough if there is a legend for the icons in the help somewhere
- if it is not too much to ask I'd love to have "Ctrl🔒¹/×2🔒" so to speak (see my previous comment)

Anyway feel free to close this bug already, AnotherTerm finally reached feature-parity with the Debian GnuRoot app: I can at last delete the latter and use your nice app instead! 8-)

green-green-avk, 2020-09-22:
>     * [minor nit:] in the settings UI I still find the key mapping sort order confusing

Just alphabetic sort order or something more button class related?

> * the "lock indicator overlay" should IMHO be much less distracting, a smallish icon to the top right like when hitting `Ctrl+I` is maybe enough if there is a legend for the icons in the help [...]

<https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv15_release>?

> Anyway feel free to close this bug already, AnotherTerm finally reached feature-parity with the GNURoot Debian app: I can at last delete the latter and use your nice app instead! relaxed

Thank you.

guillaume-d, 2020-09-24:
> > ```
> > * [minor nit:] in the settings UI I still find the key mapping sort order confusing
> > ```
> 
> Just alphabetic sort order or something more button class related?

Any sort would help, but sorting by the hardware key's label would be the most useful IMHO as said in https://github.com/green-green-avk/AnotherTerm/issues/12#issuecomment-693952385.

> ```
> * the "lock indicator overlay" should IMHO be much less distracting, a smallish icon to the top right like when hitting `Ctrl+I` is maybe enough if there is a legend for the icons in the help [...]

If only `Fn` when emulating `Alt` could behave the same as described above, or at least do not need holding before pressing the modified key, Another Term would have the best keyboard mapping ever seen on Android! 🤞 
(Sorry I did not mention that earlier, it only became apparent when using the app for real dev tasks.)

> https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv15_release?

The "lock indicator" is now much less distracting but still noticeable, I like it! 👍

green-green-avk, 2020-09-28:
BTW, feel free to leave some comment here: <https://github.com/green-green-avk/AnotherTerm/issues/5> if you have some objections. November is coming.

closed 2020-09-28 by green-green-avk