green-green-avk/AnotherTerm #13: Changing native modifier keys behavior feature

green-green-avk, 2020-09-28:
...
If only `Fn` when emulating `Alt` could behave the same as described above, or at least do not need holding before pressing the modified key, Another Term would have the best keyboard mapping ever seen on Android! 🤞 
(Sorry I did not mention that earlier, it only became apparent when using the app for real dev tasks.)
...

_Originally posted by @guillaume-d in https://github.com/green-green-avk/AnotherTerm/issues/12#issuecomment-698116885_

guillaume-d, 2020-11-12:
Would you accept a PR for this? I haven't anything yet, and it has been some time since I had a look at the code, but I think I could figure it out eventually, possibly with some guidance from you.

green-green-avk, 2020-11-13:
I see this feature rather as an additional drop down list per key entry with options like:
* "Bypass"
* "Block"
and for each <kbd>Shift</kbd>, <kbd>Alt</kbd>, <kbd>Ctrl</kbd>, <kbd>Meta</kbd> (left and right);
<kbd>Fn</kbd>, <kbd>Sym</kbd>:
* "🔒"
* "🔒¹"
* "🔓" (Maybe? To enforce no stick.)

What's your opinion?

guillaume-d, 2020-11-14:
> [...]
> Example:
> ⌨ Left Alt ➤ 💻Bypass▾ ⌨Left Alt🔒¹▾
> 
> What's your opinion?

I like the whole idea very much! 👍 But it might get pretty tight especially in portrait mode.
Don't you want to show only the locker in the second dropdown list?
Like (lame attempt at ASCII art, not all Unicode glyphs displayed correctly here): Left Alt => [Bypass `v`] [🔒¹ `v`]
Maybe that's what you meant in the first place?

green-green-avk, 2020-11-19:
1) Why not to wrap layout lines?
2) What if you want to map one key code into another?

guillaume-d, 2020-11-21:
>     1. Why not to wrap layout lines?

It could certainly help, but then would then make https://github.com/green-green-avk/AnotherTerm/issues/14 really necessary IMHO.

>     2. What if you want to map one key code into another?

I did not know you planned to implement that! 😉 
In that case your design certainly fits, contrary to mine.