green-green-avk/AnotherTerm #17: Script for linuxcontainers.org does not work for debian buster on Android 4.4

rvdginste, 2020-12-29:
Hello,

I have a tablet with Android 4.4.2 that I want to use AnotherTerm on. AnotherTerm itself seems to run fine, but I have some issues with the install-linuxcontainers script. I followed the guide on (here)[https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-under-proot.html#main_content].

I tried to install the root filesystem for debian buster (I also tried debian bulseye, ubuntu, gentoo and those all fail with a similar error; alpine does work fine however).

The first issue was that the script failed on (line 362)[https://github.com/green-green-avk/AnotherTerm-scripts/blob/2f4840fd08b0e04dc6ec6b06852f693287946413/install-linuxcontainers.sh#L362] because the folder `etc/profile.d` did not exist. This is easily remedied by adding a `mkdir -p etc/profile.d`.

The second issue is that there seems to be a problem with unpacking the root filesystem. I get the error `-25: 9: chdir() failure`. After the error, the script continues, but the terminals into the root file system immediately stop after being started. With certain root file systems I get an additional error `Pathname can't be converted from UTF-8 to current locale.`

Any ideas on what is going wrong and how it could be fixed?

thanks,
Ruben

rvdginste, 2020-12-30:
I tried AnotherTerm with the script on my phone that runs Android 8.1 and there the script runs just fine for debian. So it seems to be related to Android 4.4.2.

I was happy to find your project because it still supports KitKat, so thanks for that.

I'll try to debug the script a bit more.

rvdginste, 2020-12-31:
After further testing, I see that the problem is caused by minitar. It gives errors on hardlinks and these errors seem in turn to cause a lot of data to go missing in the unpacked root file system. This is what causes the `etc/profile.d` folder to be missing and is likely also the reason that the chroot terminal is not working.

rvdginste, 2020-12-31:
I tried to use a manipulated root file system without the hardlinks, but even with this archive, minitar does not work correctly: a lot of files are unpacked into the wrong folder.

The root file system looks like this:
```
linuxcontainers-debian-buster
\_ etc
\_ usr
      \_bin
          \_usr
               \_bin
               \_games
               \_include
               \_local
               \_src
\_var
```

So, in the root folder I only get `etc`, `usr` and `var`. I also noticed something weird with a `usr` folder nested inside `usr/bin`.

green-green-avk, 2020-12-31:
Is it an x86 phone?

rvdginste, 2021-01-01:
It's an ARMv7 phone.

green-green-avk, 2021-01-01:
Moved to <https://github.com/green-green-avk/build-libarchive-minitar-android/issues/1>.

closed 2021-01-01 by green-green-avk