green-green-avk/AnotherTerm #20: .bashrc .shrc or other

gintarasmartinaitis, 2021-09-03:
Привет всем,
в какой файл надо записывать alias? Просто у  меня так
[image]
не работает.  Прошу прощения если что не так.
С уважением Gintaras Martinaitis.

green-green-avk, 2021-09-07:
`alias` can be set for a particular shell interpreter instance and IS NOT derived by any other its instance launched from it.

A shell startup script should be used to set aliases automatically. See below.

- If it is about Android own environment (not **PRoot**ed), you can face either ksh or ash the shell interpreter. Both of them use an `ENV` environment variable to set a default startup script for each shell execution (not only the login one).

  So, just:
  1) create, to say, a `$DATA_DIR/.sh.rc`
  2) add `export ENV="$DATA_DIR/.sh.rc"` into the "execute" 
  field of your favorite:
  [image]
  3) put what you wish into the `$DATA_DIR/.sh.rc`
  4) ???
  5) Profit.

- If it is about some environment running under **PRoot**, I think, you would wish to use bash, so the only thing you need is to edit a `.bashrc` in your environment's home directory.

green-green-avk, 2023-10-27:
The main question seems resolved.

closed 2023-10-27 by green-green-avk