green-green-avk/AnotherTerm #21: "cannot access" error using ls in debian proot

dvlahakis, 2021-10-30:
Hello, thanks for the program! I am trying to start wayland via the github mk1v release. I am going through the debian install process and ran into some install issues

I get for example `ls: cannot access 'wlstart-WM': No such file or directory` when listing files under debian-bullseye proot from linuxcontainers. Works otherwise, but I'm not sure if it's something that can be fixed. This does not happen with alpine 3.14 though. This issues happens with the release version of anotherterm and the mkiv version for me.

Another issue I encountered that I'm not sure is related was that the install script for ubuntu focal from linuxcontainers fails as below:

```
Actual name: linuxcontainers-ubuntu-focal

To uninstall: run `rm -rf "$DATA_DIR/proots/linuxcontainers-ubuntu-focal"'

Creating favorites...
Done.

Arch: aarch64
Variant:
Root FS: ubuntu focal

Source: https://us.images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211018_13:48//rootfs.tar.xz

Getting Linux root FS...
20.867 MiB / 88.323 MiB of https://us.images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211018_13:48//rootfs.tar.xzPathname can't be converted from UTF-8 to current locale.
20.922 MiB / 88.323 MiB of https://us.images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211018_13:48//rootfs.tar.xz
write failed: EPIPE (Broken pipe)
Possibly URL was changed: recheck on the site.
Setting up run script...
Configuring...
./install-linuxcontainers.sh[401]: can't create bin/termsh: No such file or directory
1|:/data/data/green_green_avk.anotherterm.redist $

```
Thanks!

SilasMo, 2021-11-03:
I was getting url not resolved error and and Javax.net.ssl.SSLHandshakeException: javax.net.ssl.SSLProtocolException: SSL handshake aborted: Failure in SSL library, usually a protocol error when running the script. What i did was download an app called simple web server from Playstore, launched the webserver with all required files for the script, edited the script replacing urls in the script with the ones of the web servers and then started the script and everything ran smoothly after a long struggle. Try it maybe it will solve the error. I think the script should be edited since us.imagecontainers is not resolvable anymore world wide(tested on varius websites).

dvlahakis, 2021-11-03:
Thanks for clarifying the problem! I'll try your solution.

SilasMo, 2021-11-03:
By the way when i first edited the script, I tried to copy the files using `cp` comand to the root environment of the app and the proot root environment but the error `./install-linuxcontainers.sh[382]: can't create bin/termsh: No such file or directory` appeared. After doing all i stated above everything ran nicely. So there is a high chance that everything will get solved since we have similar errors

green-green-avk, 2021-11-05:
Yep, `us.images.linuxcontainers.org` is no more => `images.linuxcontainers.org`.

* [x] <https://github.com/green-green-avk/AnotherTerm-scripts/commit/6ccd91ac5390421b5465d2288f3649fc741414ec>
* [x] <https://github.com/green-green-avk/AnotherTerm-docs/commit/f74bf0da89cbbef7c6e7b4e2edae335ed6ac3542>
* [x] `small-proxy-301517` on **Google Cloud** updated.

Things should work good now. Please, confirm.

## PS:
> I was getting url not resolved error and and `Javax.net.ssl.SSLHandshakeException: javax.net.ssl.SSLProtocolException: SSL handshake aborted: Failure in SSL library, usually a protocol error`...

Yep... See <https://green-green-avk.github.io/AnotherTerm-docs/issues.html#termsh-ssl-handshake-error-on-very-old-androids>

SilasMo, 2021-11-05:
After downloading the app from google play, i was blinded. Then i came and read the docs and was enlightened. But still i didnt get the point because im my head i had told my self that android 4.4.4 is not too old :D.
In the install-linuxcontainers.sh script, i had replaced `us.images.linuxcontainers.org` with `images.linuxcontainers.org` but still received the java net ssl error, so i had no option but to opt to another method.

dvlahakis, 2021-11-05:
Hi,  I get the below when I run the script:

( S=repo-linuxcontainers.sh ; "$TERMSH" copy -f -fu "https://raw.githubusercontent.com/green-green-avk/AnotherTerm-scripts/master/$S" -tp . && chmod 755 $S && sh ./$S )

Actual name: linuxcontainers-ubuntu-focal

To uninstall: run `rm -rf "$DATA_DIR/proots/linuxcontainers-ubuntu-focal"'

Creating favorites...
Done.

Arch: aarch64
Variant:
Root FS: ubuntu focal

Source: https://images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211105_07:42//rootfs.tar.xz

Getting Linux root FS...
19.164 MiB / 88.895 MiB of https://images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211105_07:42//rootfs.tar.xzPathname can't be converted from UTF-8 to current locale.
19.531 MiB / 88.895 MiB of https://images.linuxcontainers.org//images/ubuntu/focal/arm64/default/20211105_07:42//rootfs.tar.xz
write failed: EPIPE (Broken pipe)
Possibly URL was changed: recheck on the site.
Setting up run script...
Configuring...
./install-linuxcontainers.sh[401]: can't create bin/termsh: No such file or directory
1|:/data/data/green_green_avk.anotherterm.redist $

Thanks!

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

green-green-avk, 2021-11-05:
Hmm...
Reproduced.
* **Ubuntu Focal** fails.
* **Debian Bullseye** works.

It looks like a [`minitar`](https://github.com/green-green-avk/build-libarchive-minitar-android) (or its dependencies) bug.
Digging...

SilasMo, 2021-11-06:
Why is you output different here is mine when installing Debian Buster
```
u0_a176@goyave3g:/data/data/green_green_avk.anotherterm.oldgood $ cd /storage/emulated/0/Download
u0_a176@goyave3g:/storage/emulated/0/Download $ ./install-linuxcontainers.sh Debian Buster

Installation subdir name proots/___ [linuxcontainers-Debian-Buster]: Buster

Actual name: Buster

To uninstall: run `rm -rf "$DATA_DIR/proots/Buster"'

Creating favorites...
Done.

Arch: armv7a
Variant: -pre5
Root FS: Debian Buster

Source: http://192.168.43.239:8080/rootfs.tar.xz

Getting minitar...
Getting PRoot...
Getting Linux root FS...

/etc/passwd:
=======
root:x:0:0:root:/root:/bin/bash
daemon:x:1:1:daemon:/usr/sbin:/usr/sbin/nologin
bin:x:2:2:bin:/bin:/usr/sbin/nologin
sys:x:3:3:sys:/dev:/usr/sbin/nologin
sync:x:4:65534:sync:/bin:/bin/sync
games:x:5:60:games:/usr/games:/usr/sbin/nologin
man:x:6:12:man:/var/cache/man:/usr/sbin/nologin
lp:x:7:7:lp:/var/spool/lpd:/usr/sbin/nologin
mail:x:8:8:mail:/var/mail:/usr/sbin/nologin
news:x:9:9:news:/var/spool/news:/usr/sbin/nologin
uucp:x:10:10:uucp:/var/spool/uucp:/usr/sbin/nologin
proxy:x:13:13:proxy:/bin:/usr/sbin/nologin
www-data:x:33:33:www-data:/var/www:/usr/sbin/nologin
backup:x:34:34:backup:/var/backups:/usr/sbin/nologin
list:x:38:38:Mailing List Manager:/var/list:/usr/sbin/nologin
irc:x:39:39:ircd:/var/run/ircd:/usr/sbin/nologin
gnats:x:41:41:Gnats Bug-Reporting System (admin):/var/lib/gnats:/usr/sbin/nologin
nobody:x:65534:65534:nobody:/nonexistent:/usr/sbin/nologin
_apt:x:100:65534::/nonexistent:/usr/sbin/nologin
systemd-timesync:x:101:102:systemd Time Synchronization,,,:/run/systemd:/usr/sbin/nologin
systemd-network:x:102:103:systemd Network Management,,,:/run/systemd:/usr/sbin/nologin
systemd-resolve:x:103:104:systemd Resolver,,,:/run/systemd:/usr/sbin/nologin
messagebus:x:104:105::/nonexistent:/usr/sbin/nologin
=======
Regular user name [my_acct]: my_acct
Preferred shell [/bin/bash]: /bin/bash

Setting up run script...
Configuring...

Done!

u0_a176@goyave3g:/storage/emulated/0/Download $
```
> It looks like a minitar (or its dependencies) bug.
Digging...

In my output, minitar and proot has to be first fetched. But in his output i dont this minitar was fetched nicely because there was no echo of getting minitar or proot.

SilasMo, 2021-11-06:
@dvlahakis  try clearing your app data and starting afresh. To me if seems like the plugkns are broken.

green-green-avk, 2021-11-06:
> In my output, minitar and proot has to be first fetched. But in his output i dont think minitar was fetched nicely because there was no echo of getting minitar or proot.

It seems, @dvlahakis has <https://github.com/green-green-avk/AnotherTermShellPlugin-Android10Essentials> installed.
The script automatically uses plugin-provided utils if available.

dvlahakis, 2021-11-06:
I think I do have the plug in installed. However, I also have installed Debian successfully,  it is ubuntu that fails.

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

SilasMo, 2021-11-06:
ohhh i get it now

green-green-avk, 2021-11-07:
<https://github.com/green-green-avk/build-libarchive-minitar-android/issues/2#issuecomment-962553485>

green-green-avk, 2021-11-07:
<https://github.com/green-green-avk/AnotherTermShellPlugin-Android10Essentials/releases/tag/v1.7_release>

green-green-avk, 2021-11-07:
Please, verify.

green-green-avk, 2021-11-07:
* <https://github.com/green-green-avk/AnotherTerm-scripts/commit/e1ece551c102b1f546f1947d86725e4c10d8e796>

Please, verify now.

green-green-avk, 2021-11-07:
* <https://github.com/green-green-avk/AnotherTerm-scripts/commit/e1ece551c102b1f546f1947d86725e4c10d8e796>
* <https://github.com/green-green-avk/AnotherTerm-scripts/commit/5f176aed008b55197ba450a96a344afebb86d570>

Please, verify now.

dvlahakis, 2021-11-07:
Great! The latest patch you just made fixed the resolv.conf error I was having. I just installed ubuntu focal and it seems to have installed and dns is working. Thanks!

closed 2021-11-10 by green-green-avk