green-green-avk/AnotherTerm #24: Question regarding shared memory Android 11 in debian proot

dvlahakis, 2021-11-08:
Hi. I'm trying to install wayland support on a debian bullseye proot. I have the mkiv beta latest release.

I have installed the scripts according to the instructions I believe. First libwrapdroid and then I run the commands:

```
export LIBWRAPDROID_SOCKET_NAME=mysocket
export LIBWRAPDROID_AUTH_KEY=1234567890987654 # not less than 16 hexidecimal digits
export LD_PRELOAD="/opt/shm/lib/libwrapdroid-shm-sysv.so:/opt/shm/lib/libwrapdroid-shm-posix.so${LD_PRELOAD:+:$LD_PRELOAD}"
/opt/shm/bin/libwrapdroid-server & #because I need to background this as the command does not return to prompt

```

Then I run the execute commands listed:
```
/system/bin/sh \
"$DATA_DIR/proots/linuxcontainers-debian-buster/run" \
0:0 ./wlstart-X | \
/system/bin/sh \
"$DATA_DIR/proots/linuxcontainers-debian-buster/run" \
'' ./wlstart-WM
```
and I get the following output

`CANNOT LINK EXECUTABLE "CANNOT LINK EXECUTABLE "/system/bin/sh/system/bin/sh": ": library "libpthread.so.0" not found: needed by /opt/shm/lib/libwrapdroid-shm-sysv.so in namespace (default)library "libpthread.so.0" not found: needed by /opt/shm/lib/libwrapdroid-shm-sysv.so in namespace (default)
`

I also did an ls to see if libpthread exists, and I get the below, which is also unclear to me because it says it is not there but it is. Actually, that is another issue for the proot, but I'm not sure it is related. I get this error for all the files in /lib. This doesn't happen for ubuntu focal proot:

`ls: cannot access '/lib/aarch64-linux-gnu/libpthread.so': No such file or directory`

dvlahakis, 2021-11-08:
Forgot to say thank you for your app!

green-green-avk, 2021-11-08:
It looks like you messed up with run environment of
```sh
/system/bin/sh \
"$DATA_DIR/proots/linuxcontainers-debian-buster/run" \
0:0 ./wlstart-X | \
/system/bin/sh \
"$DATA_DIR/proots/linuxcontainers-debian-buster/run" \
'' ./wlstart-WM
```
It must be run in the Android own shell and start all the other scripts by itself (in order to sync the security tokens).

Please, see [updated](https://github.com/green-green-avk/AnotherTerm-docs/commit/a64c2c5111b3377f7f81edfa21ff599659136681#diff-46b42b4229cd7a39c564e780bb665a8bde4fdf722007e8473f167fe53ed4b995)
<https://green-green-avk.github.io/AnotherTerm-docs/graphical-sessions.html#main_content>

green-green-avk, 2021-11-08:
Please, also take into account this workaround: <https://github.com/green-green-avk/AnotherTerm/issues/25>.

green-green-avk, 2021-11-09:
And this workaround: <https://github.com/Neo-Oli/termux-ubuntu/issues/35#issuecomment-636368078>.

dvlahakis, 2021-11-09:
Thanks for the update! I have created a new terminal with the execute command thanks to your shortcut. I'm still not sure where to put the shared memory api commands below:

`export LIBWRAPDROID_SOCKET_NAME=mysocket
export LIBWRAPDROID_AUTH_KEY=1234567890987654 # not less than 16 hexidecimal digits
export LD_PRELOAD="/opt/shm/lib/libwrapdroid-shm-sysv.so:/opt/shm/lib/libwrapdroid-shm-posix.so${LD_PRELOAD:+:$LD_PRELOAD}"
/opt/shm/bin/libwrapdroid-server`

I tried executing the statements in a debian terminal session  but I have no clue if that is correct or has the intended effect.  I get the following error when I subsequently run a second separate graphical debian session (with the execute setting):

`/usr/bin/startxfce4: X server already running on display :0
(process:13778): xfce4-session-CRITICAL **: 06:56:41.553: dbus-launch not found, the desktop will not work properly!`

Thanks for your time!

green-green-avk, 2021-11-09:
> ...I'm still not sure where to put the shared memory api commands below...

It is already in the start-up scripts: <https://green-green-avk.github.io/AnotherTerm-docs/graphical-sessions.html#script-startwm>

> ...error when I subsequently run a second separate graphical debian session...

* When the script actually exited?
* Did it open a graphical session at least?

dvlahakis, 2021-11-10:
Hi, that error happens but the script does not appear to exit, it just hangs. No graphical session is started. Thanks

dvlahakis, 2021-11-10:
Hi, the script in that case just halted and showed that result. No graphical session started. 
I then installed debian buster, which does not have the directory listing errors bullseye had. I also added the shared memory patch and session for buster, and added the three scripts to root and my_acct and made them executable. I also installed xwayland, python3 and xfce4. Then when I run the graphical session I get:

```
usr/bin/startxfce4: X server already running on display :0
_IceTransmkdir: ERROR: euid != 0,directory /tmp/.ICE-unix will not be created.
gpg-agent[32455]: WARNING: "--write-env-file" is an obsolete option - it has no effect
gpg-agent[32455]: directory '/home/my_acct/.gnupg' created
gpg-agent[32455]: directory '/home/my_acct/.gnupg/private-keys-v1.d' created
gpg-agent[32456]: gpg-agent (GnuPG) 2.2.12 started

(xfce4-session:32441): xfce4-session-WARNING **: 07:46:05.531: gpg-agent returned no PID in the variables

(xfce4-session:32441): xfce4-session-WARNING **: 07:46:05.534: xfsm_manager_load_session: Something wrong with /home/my_acct/.cache/sessions/xfce4-session-localhost:0, Does it exist? Permissions issue?
Connection failure: Connection refused
pa_context_connect() failed: Connection refused

(xfsettingsd:32476): libupower-glib-WARNING **: 07:46:05.719: Couldn't connect to proxy: Could not connect: No such file or directory

(xfsettingsd:32476): libupower-glib-CRITICAL **: 07:46:05.720: up_client_get_lid_is_closed: assertion 'UP_IS_CLIENT (client)' failed

(xfsettingsd:32476): GLib-GObject-WARNING **: 07:46:05.720: invalid (NULL) pointer instance

(xfsettingsd:32476): GLib-GObject-CRITICAL **: 07:46:05.721: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(xfsettingsd:32476): GLib-CRITICAL **: 07:46:05.805: g_str_has_prefix: assertion 'prefix != NULL' failed

(xfsettingsd:32476): GLib-GObject-CRITICAL **: 07:46:05.807: g_value_get_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed

(xfsettingsd:32476): GLib-GObject-CRITICAL **: 07:46:05.818: g_value_get_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed

(polkit-gnome-authentication-agent-1:32477): polkit-gnome-1-WARNING **: 07:46:05.944: Error getting authority: Error initializing authority: Could not connect: No such file or directory

** (light-locker:32473): ERROR **: 07:46:05.946: session_id is not set, is /proc mounted with hidepid>0?

(xfwm4:32458): GLib-CRITICAL **: 07:46:07.625: g_str_has_prefix: assertion 'prefix != NULL' failed

(xfwm4:32458): xfwm4-WARNING **: 07:46:07.659: The property '/general/double_click_distance' of type int is not supported

(xfsettingsd:32476): xfsettingsd-WARNING **: 07:46:07.672: Failed to get the _NET_NUMBER_OF_DESKTOPS property.

(xfwm4:32458): xfwm4-WARNING **: 07:46:07.720: Error opening /dev/dri/card0: Permission denied

(xfdesktop:32466): GVFS-RemoteVolumeMonitor-WARNING **: 07:46:08.060: remote volume monitor with dbus name org.gtk.vfs.UDisks2VolumeMonitor is not supported

(xfwm4:32458): xfwm4-WARNING **: 07:46:10.476: Another compositing manager is running on screen 0
```

and it freezes there.

green-green-avk, 2021-11-10:
So, when you are going back to the sessions list, you see no graphical session in it??? The log looks like it is perfectly connected to the Wayland backend...

dvlahakis, 2021-11-10:
I'm sorry I didn't realize the session would appear separately, and I was able to go into the graphical session successfully! Below is the session log:

```
/usr/bin/startxfce4: X server already running on display :0
_IceTransmkdir: Owner of /tmp/.ICE-unix should be set to root
gpg-agent[9831]: WARNING: "--write-env-file" is an obsolete option - it has no effect
gpg-agent[9832]: gpg-agent (GnuPG) 2.2.12 started

(xfce4-session:9812): xfce4-session-WARNING **: 16:24:15.044: gpg-agent returned no PID in the variables

(xfce4-session:9812): xfce4-session-WARNING **: 16:24:15.048: xfsm_manager_load_session: Something wrong with /home/my_acct/.cache/sessions/xfce4-session-localhost:0, Does it exist? Permissions
issue?
Connection failure: Connection refused
pa_context_connect() failed: Connection refused

(xfce4-session:9812): xfce4-session-WARNING **: 16:24:15.224: I/O error opening ICE connection 0x3000097520

(polkit-gnome-authentication-agent-1:9855): polkit-gnome-1-WARNING **: 16:24:15.460: Error getting authority: Error initializing authority: Could not connect: No such file or directory

** (light-locker:9850): ERROR **: 16:24:15.469: session_id is not set, is /proc mounted with hidepid>0?

(xfwm4:9835): xfwm4-WARNING **: 16:24:15.982: Error opening /dev/dri/card0: Permission denied
xfce4-panel: No window manager registered on screen 0. To start the panel without this check, run with --disable-wm-check.

(xfce4-session:9812): xfce4-session-WARNING **: 16:24:46.617: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or directory)

(xfce4-session:9812): xfce4-session-WARNING **: 16:24:46.619: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or directory)
Couldn't find a suitable web browser!
Set the BROWSER environment variable to your desired browser.

(xfce4-session:9812): xfce4-session-WARNING **: 16:27:52.269: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or directory)

(xfce4-session:9812): xfce4-session-WARNING **: 16:27:52.275: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or directory)
(EE) failed to read Wayland events: Broken pipe
xfce4-appfinder: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.
xfce4-session: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
xfce4-panel: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.
wrapper-1.0: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.
wrapper-1.0: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.
gpg-agent[9832]: SIGTERM received - shutting down ...
gpg-agent[9832]: gpg-agent (GnuPG) 2.2.12 stopped
stopping X at PID 9779...
./wlstart-WM: line 12: kill: (9779) - No such process

```

I was not able to execute anything, maybe I need to install more of the xfce environment via tasksel so it can find everything. I also was not able to type into any dialog boxes, but that might be a result of the missing programs? I think the log indicates pm-is-supported is missing, I can look that up. I'm using Samsung Dex, and the window works perfectly on the attached screen. Resizes quickly and correctly on my limited testing. So the only issues I see at this point are missing programs and can't type.

dvlahakis, 2021-11-10:
Ok, running apt install task-xfce-desktop did the trick to populate the programs. There is now a quirk, there are two mouse cursors showing, which is more annoying than a problem. Also there is still a keyboard issue. Backspace and cursor keys don't work on my keyboard but they do work on the built in anotherterm onscreen keyboard. So that makes desktop use problematic. But thanks because I definitely appreciate the potential!

green-green-avk, 2021-11-10:
Uh huh... Vendor's own external keyboard mapping ~bugs~ features (or maybe even keyboard own improper HID descriptors). Please, try different USB and Bluetooth keyboards - some could work.

Please, also take into account this one: <https://github.com/green-green-avk/AnotherTerm/issues/26>
Some work on the screen keyboard is still to be done.

green-green-avk, 2021-11-10:
And this one: <https://github.com/green-green-avk/AnotherTerm/issues/27>

dvlahakis, 2021-11-11:
Tried another keyboard, a nexdock 360 connected via usb, but the backspace key  did not work.  I will try more if I get a chance. I also tried to see if remapping keys might help, but could not figure that out. CTRL-H does work for backspace, and ctrl-B,F for cursor. So hardware keyboards can work with workarounds.

dvlahakis, 2021-11-11:
Addendum to above,,,,,, I was able to start tigervnc scraping server on the buster session, and connect to that vnc session after starting the graphical wayland session. I connected using the client avnc. I was able to use the backspace and arrow keys in the vnc session which mirrored the wayland session. Wayland still didn't work. Thanks!

green-green-avk, 2021-11-11:
Did backspace and cursors not work in any screen keyboard mode (Off, Built-in, IME)?

dvlahakis, 2021-11-11:
I can't get them to work when the anotherterm built in soft keyboard is active or when it is toggled off. I am not sure how to have the system soft keyboard pop up when the hardware keyboard is attached so I don't know for sure, I have to look further.

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

green-green-avk, 2021-11-11:
Sorry, I forgot that IME cannot be shown when the hardware keyboard is attached.
Funny: I blocked this combination as a workaround for non-working hardware keyboard <kbd>Backspace</kbd> in older Androids...

dvlahakis, 2021-11-12:
I had also run xev from a terminal because I was trying to see if the keys were registering under the graphical Wayland session. The backspace and cursor keys did not register anything at all for xev. So remapping was not an option I guess.

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

Funny: I blocked this combination as a workaround for non-working hardware keyboard Backspace in older Androids...

green-green-avk, 2021-11-12:
By the way, do the backspace and cursor keys work in the ANSI terminal sessions?

dvlahakis, 2021-11-12:
Yes, they work normally in the default anotherterm terminal sessions.

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

green-green-avk, 2021-11-12:
Oops... I see it too.

Thank you, for providing me with some inspiration to finish the work.

One more question: do you prefer to see several keyboards as either separate devices (in terms of Wayland protocol) or one aggregate device?

dvlahakis, 2021-11-12:
Thanks for the app! I appreciate the effort and the result.
I'm not sure I understand,  do you mean separate keyboards in the Wayland proot environment? I often switch from hardware to soft keyboards, because I move around for work, so I appreciate being able to easily pick the one that works best at the moment. I hope that answers your question.

Dimitri Vlahakis
e. ***@***.***
c. +19174187640

green-green-avk, 2021-11-12:
Ok. I'll try to do it in the best possible way.

dvlahakis, 2021-11-13:
Hello. I just installed the graphical window in my debian bullseye session, and it is working very well. I was able to run two graphical sessions, which is a great bonus! 

I can confirm the backspace and cursors do not work in bullseye also though. 

Also as an aside in case there were edit notifications I removed my phone number which I noticed in prior comments I made, sorry. Thanks again!

green-green-avk, 2022-02-02:
[<kbd>MkIV-dev16</kbd>](https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIV_dev16_release)
<https://github.com/green-green-avk/AnotherTerm-scripts/blob/master/Xwayland/README.md>

* <https://github.com/green-green-avk/AnotherTerm/issues/26> - hardware keyboard support is also fixed there
* <https://github.com/green-green-avk/AnotherTerm/issues/27>

Let's try it.

dvlahakis, 2022-02-02:
Hello. I tried to execute the debian buster and bullseye wm sessions I used before to test the keyboard, but I get the following error log where it used to work (My samsung has updated to Android 12 in case that might have an effect). Any suggestions as to what I am doing wrong?

`/usr/bin/startxfce4: X server already running on display :0
_IceTransmkdir: Owner of /tmp/.ICE-unix should be set to root
gpg-agent[11699]: WARNING: "--write-env-file" is an obsolete option - it has no effect
gpg-agent[11701]: gpg-agent (GnuPG) 2.2.12 started

(xfce4-session:11687): xfce4-session-WARNING **: 06:45:14.385: gpg-agent returned no PID in the variables

(xfce4-session:11687): xfce4-session-WARNING **: 06:45:14.388: xfsm_manager_load_session: Something wrong with /home/my_acct/.cache/sessions/xfce4-session-localh
ost:0, Does it exist? Permissions issue?
Connection failure: Connection refused
pa_context_connect() failed: Connection refused

(xfsettingsd:11718): libupower-glib-WARNING **: 06:45:14.812: Couldn't connect to proxy: Could not connect: No such file or directory

(xfsettingsd:11718): libupower-glib-CRITICAL **: 06:45:14.812: up_client_get_lid_is_closed: assertion 'UP_IS_CLIENT (client)' failed

(xfsettingsd:11718): GLib-GObject-WARNING **: 06:45:14.815: invalid (NULL) pointer instance

(xfsettingsd:11718): GLib-GObject-CRITICAL **: 06:45:14.821: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(polkit-gnome-authentication-agent-1:11743): polkit-gnome-1-WARNING **: 06:45:14.942: Error getting authority: Error initializing authority: Could not connect: N
o such file or directory

** (light-locker:11716): ERROR **: 06:45:14.955: session_id is not set, is /proc mounted with hidepid>0?
/usr/share/system-config-printer/applet.py:27: DeprecationWarning: Importing dbus.glib to use the GLib main loop with dbus-python is deprecated.
Instead, use this sequence:

    from dbus.mainloop.glib import DBusGMainLoop

    DBusGMainLoop(set_as_default=True)

  import dbus.glib

(nm-applet:11748): nm-applet-WARNING **: 06:45:15.134: Error connecting to system D-Bus: Could not connect: No such file or directory

(nm-applet:11748): nm-applet-WARNING **: 06:45:15.136: Could not connect: No such file or directory

(xfsettingsd:11718): xfsettingsd-WARNING **: 06:45:15.318: Failed to get the _NET_NUMBER_OF_DESKTOPS property.
/usr/share/system-config-printer/applet.py:45: PyGIWarning: Notify was imported without specifying a version first. Use gi.require_version('Notify', '0.7') befor
e import to ensure that the right version gets loaded.
  from gi.repository import Notify
system-config-printer-applet: failed to connect to system D-Bus

(xfdesktop:11711): GVFS-RemoteVolumeMonitor-WARNING **: 06:45:15.715: remote volume monitor with dbus name org.gtk.vfs.UDisks2VolumeMonitor is not supported

(xfwm4:11703): xfwm4-WARNING **: 06:45:15.823: Error opening /dev/dri/card0: Permission denied

(xfce4-session:11687): xfce4-session-WARNING **: 06:45:15.980: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or
directory)

(xfce4-session:11687): xfce4-session-WARNING **: 06:45:15.985: failed to run script: Failed to execute child process “/usr/bin/pm-is-supported” (No such file or
directory)

(nm-applet:11748): libnm-CRITICAL **: 06:45:16.070: ((libnm/nm-client.c:271)): assertion '<dropped>' failed

(nm-applet:11748): libnm-CRITICAL **: 06:45:16.071: ((libnm/nm-client.c:232)): assertion '<dropped>' failed

(nm-applet:11748): libnm-CRITICAL **: 06:45:16.428: ((libnm/nm-client.c:989)): assertion '<dropped>' failed
(EE)
(EE) Backtrace:
`

dvlahakis, 2022-04-23:
Hello. I'm getting the following error when running the graphical session (debian bullseye). Νοt sure I followed the instructions correctly for libwrapdroid or if something needs to be done, but regardless this is the output i get upon starting the session. Thanks!

Traceback (most recent call last)
  File "/root/./wlstart-X", line 42, in <module>
    sock = connect(wl_sock)
  File "/root/./wlstart-X", line 33, in connect
    sock.connect(addr)
ConnectionRefusedError: [Errno 111] Connection refused
=! LIBWRAPDROID_AUTH_KEY is not set!
/usr/bin/startxfce4: X server already running on display :
xrdb: Can't open display ':'
Helper: Starting...
Helper: Error: [Errno 111] Connection refused
Unable to init server: Could not connect: Connection refused
xfce4-session: Cannot open display: .
Type 'xfce4-session --help' for usage.
/home/my_acct/startwm: line 16: kill: (17871) - No such process
/home/my_acct/startwm: line 17: kill: (17870) - No such process
stopping X at PID ...
./wlstart-WM: line 12: kill: `': not a pid or valid job spec

green-green-avk, 2022-05-04:
Hmm...
A connection to the built-in Wayland server seems refused in `/root/wlstart-X` so all other things are broken after it.
What is your Android version? I'm going to recreate this situation...

dvlahakis, 2022-05-04:
Hi! It is android 12. One ui 4.1 on a galaxy note 20. Thanks!

green-green-avk, 2022-05-05:
Did Wayland stop working after a system update?

green-green-avk, 2022-05-05:
And try to run `termsh` without any arguments in your PRooted environment: will it show a help message or not?

dvlahakis, 2022-05-06:
Wayland stopped working after the update to oneui 4 on my system, which was I guess the switch to android 12. To be fair I had not run it for a little while, but I have not made any other major changes. I tried reinstalling the proot and wm, but that did not solve the problem. I just ran termsh and it does display a long help message after the following error:

WARNING: linker: Warning: failed to find generated linker configuration from "/linkerconfig/ld.config.txt"

Thanks!

green-green-avk, 2022-06-01:
Hmm... Your Android 12 even does not have a `dup` syscall blocked by the **Google** SELinux rules...
Anyway, they blocked not all of the three `dup`s, so it will be solved (I hope 🤞)...

dvlahakis, 2022-06-02:
The graphical window is working for me now. I am using a debian buster container, and typing this in firefox. I'm using Samsung Dex and , and the window is sizing nicely and is stable.
I'm not sure if there was a change that fixed it or if it was a mistake I was making. I was installing the MKiv and Mkiii versions side by side, but I think at some point the versions overwrote each other and I was using the mkiii version, which I believe does not have the graphical capability? I made sure the correct version was running and it is working well right now.
Thanks!

green-green-avk, 2022-07-04:
Which flavor do you use? I recommend <kbd>oldgood</kbd> as it targets **Android&nbsp;9** and thus declares older and less strict compatibility rules.

dvlahakis, 2022-07-04:
I am using the redist flavor on my samsung note 20 ultra. Should I install the oldgood flavor over it?

green-green-avk, 2022-08-05:
Yes. Changing flavors is reversible.

dvlahakis, 2022-08-05:
Got it. Thanks!