green-green-avk/AnotherTerm #25: PRooted Debian: aptitude fails to install a package

green-green-avk, 2021-11-08:
```
root@localhost:~# aptitude install xwayland
[ ERR] Writing extended state information
[ ERR] Building tag database
E: failed to rename /var/lib/aptitude/pkgstates to /var/lib/aptitude/pkgstates.old - save_selection_list (1: Operation not permitted)
```

green-green-avk, 2021-11-08:
Workaround:
```sh
touch /var/lib/aptitude/pkgstates
```

Yes, this simple.

dvlahakis, 2021-11-09:
Apt seems to work, and is able to install aptitude. However, aptitude complains with lock file errors. 

`│E: Could not open lock file /var/lock/aptitude - open (2: No such file or directory)                                                                                                           ▒│
│E: Could not initialize dependency cache                                                                                                                                                       ▒│
│W: Could not lock the cache file; this usually means that dpkg or another apt tool is already installing packages.  Opening in read-only mode; any changes you make to the states of packages  ▒│
│   will NOT be preserved!`

When I Is /var, I get the following:

```
ls: cannot access 'mail': No such file or directory
ls: cannot access 'log': No such file or directory
ls: cannot access 'backups': No such file or directory
ls: cannot access 'tmp': No such file or directory
ls: cannot access 'local': No such file or directory
ls: cannot access 'opt': No such file or directory
ls: cannot access 'lock': No such file or directory
ls: cannot access 'lib': No such file or directory
ls: cannot access 'spool': No such file or directory
ls: cannot access 'run': No such file or directory
total 16
drwxr-xr-x. 31 root root 4096 Dec 31  2008 .
drwxr-xr-x. 31 root root 4096 Dec 31  2008 ..
d??????????  ? ?    ?       ?            ? backups
drwxrwx---.  8 1000 2001 4096 Sep 24  2019 cache
d??????????  ? ?    ?       ?            ? lib
d??????????  ? ?    ?       ?            ? local
l??????????  ? ?    ?       ?            ? lock
d??????????  ? ?    ?       ?            ? log
d??????????  ? ?    ?       ?            ? mail
d??????????  ? ?    ?       ?            ? opt
l??????????  ? ?    ?       ?            ? run
d??????????  ? ?    ?       ?            ? spool
d??????????  ? ?    ?       ?            ? tmp
```

green-green-avk, 2021-11-09:
What is your
```sh
cat /etc/proot/run.cfg
```
?

dvlahakis, 2021-11-10:
Hi. the run.cfg for debian bullseye is

`# Main configuration

# Regular user name
USER=my_acct

# Preferred shell (fallback: /bin/sh)
SHELL=/bin/bash

# =======

PROOT=$("$TERMSH" plugin 'green_green_avk.anothertermshellplugin_android10essentials' proot)
PROOT_USERLAND=$("$TERMSH" plugin 'green_green_avk.anothertermshellplugin_android10essentials' proot-userland)

# Mostly for Android < 5 now. Feel free to adjust.
# Not recommended to set it >= '4.8.0' for kernels < '4.8.0'
# becouse of a random number generation API change at this point
# as it could break libopenssl random number generation routine.
_KERNEL_VERSION="4.19.113-22701397"
if [ "${_KERNEL_VERSION%%.*}" -lt 4 ] ; then
 PROOT_OPT_ARGS+=('-k' '4.0.0')
fi

# Android >= 9 can have a read restictiction
# on '/proc/version'.
cat /proc/version >/dev/null 2>&1 || {
 _PROC_VERSION="$CFG_DIR/proc.version.cfg"
 { uname -a 2>/dev/null || echo 'Linux' ; } > "$_PROC_VERSION"
 PROOT_OPT_ARGS+=('-b' "$_PROC_VERSION:/proc/version")
}

# Application data shared directory.
PROOT_OPT_ARGS+=('-b' "$SHARED_DATA_DIR:/mnt/shared")

# Uncomment to manipulate Android application own private data directory.
#PROOT_OPT_ARGS+=('-b' '/data')

# =======`

green-green-avk, 2021-11-10:
Hello, `statx()` syscall... I'll try to fix **PRoot** promptly.

dvlahakis, 2021-11-10:
Thank you. I guess that would be in an apk update?

green-green-avk, 2021-11-12:
Ready:
<https://github.com/green-green-avk/build-proot-android/commit/72ef4646a8930ff795e86fbcf7cf77e46359530e>
<https://github.com/green-green-avk/AnotherTermShellPlugin-Android10Essentials/releases/tag/v1.8_release>

dvlahakis, 2021-11-12:
Seems to fix the problem, I can run ls in the bullseye session and I get good listings. Thanks!

nathansstm, 2024-08-13:
I am very happy I found this Ty!!!