green-green-avk/AnotherTerm #26: Graphical session screen keyboard

green-green-avk, 2021-11-10:
It needs to be fully implemented.
It seems it should look like a set of switchable keyboards:
* Text input ANSI terminal like keyboard: each symbol generates a couple of down/up events surrounded by appropriate modifier keys down/up events.
* Action keyboard: down/up events are emitted as keys pressed/released and thus no gesture input.
* Android IME: behavior is similar to text input ANSI terminal like keyboard.

green-green-avk, 2021-11-10:
To do:
- [ ] Split current screen keyboard view into Graphical terminal and ANSI terminal ones in order to isolate possible bugs during development.
- [ ] Split current Graphical terminal screen keyboard into **text input** and **action** flavours.
- [ ] Finish implementation of the **text input** keyboard.
- [ ] Do possible corrections to the **Android IME** keyboard.
- [ ] Finish implementation of the **action** keyboard.

green-green-avk, 2021-11-12:
## Problem
1. Wayland protocol actually provides input method (text input) / clipboard support.
2. `Xwayland` does not implement it.
3. I hate perpetual patching / recompilation.

## Solution
- [ ] Add a small custom protocol extension to provide an additional socket descriptor to a dedicated helper.

green-green-avk, 2022-01-11:
Split ANSI/Graphics: <https://github.com/green-green-avk/AnotherTerm/commit/f26c026cf09dd47b380dd706b6da3da5eb2af82e>

green-green-avk, 2022-01-31:
Input method: <https://github.com/green-green-avk/AnotherTerm/commit/a639cfa9959e06f25c40ef886f28a54ecd249e8b>

green-green-avk, 2022-01-31:
Rework:
* <https://github.com/green-green-avk/AnotherTerm/commit/38a74a6505af0049fb5a51f09a2773ffe6087924>
* <https://github.com/green-green-avk/AnotherTerm/commit/ecdd0a1b5610ea74c4882b6799c1d269cbd5b937>
* <https://github.com/green-green-avk/AnotherTerm/commit/4f8d4d8ba295f98dc0dd5f90a5a7fec05c9b1a32>

green-green-avk, 2022-02-02:
Done.

Usage of Android international keyboard input and clipboard integration:
<https://github.com/green-green-avk/AnotherTerm-scripts/tree/master/Xwayland>

For further development see <https://github.com/green-green-avk/AnotherTerm/issues/28>

closed 2022-02-02 by green-green-avk

UMERFARMAN, 2022-02-22:
> It needs to be fully implemented.
> It seems, it should look like a set of switchable keyboards:
> * **Text input** ANSI terminal like keyboard: each symbol generates a couple of down/up events surrounded by appropriate modifier keys down/up events.
> * **Action** keyboard: down/up events are emitted as keys pressed/released and thus no gesture input is possible.
> * **Android IME**: behavior is similar to text input ANSI terminal like keyboard.