green-green-avk/AnotherTerm #30: versionCode and releases

IzzySoft, 2022-06-11:
It seems that since MkIIIv34, `versionCode` was never increased: it's still stating `90`. As Android uses this to tell versions apart, new releases are never announced to the user.

Further, the current tag naming makes it hard to automatically fetch updates (I just noticed today that I missed quite a few of them, as `MkIIIv34` compared against `MkIV_dev20_release` was interpreted as `34 > 20` and thus the new tag being ignored). Can you please clarify the difference between MkIII and MkIV? Looks like these are two "branches" released at the same time.

For fetching updates it would help enormously having tag names with clear versioning as well as corresponding `versionName` or `versionCode` (you could e.g. use prefixes as you currently seem to do: `mkiii_<versionCode>` and `mkiv_<versionCode>` would be great as that way you cannot forget increasing it either :wink:)

Thanks in advance!

green-green-avk, 2022-07-04:
It seems, `MkIIIv` and `MkIV-dev` version prefixes are pretty good to represent both branches now. I'm not planning to remove the `-dev` suffix soon. I hope, you have a way to publish two applications with the same **ID** as it makes interchangeability between <kbd>Wayland</kbd> and <kbd>stable</kbd> (master branch) editions much simpler.

IzzySoft, 2022-07-04:
No, applicationId is unique per app. While using the same applicationId would make cross-updates easier, it would also make **accidental** cross-updates easier – as Android wouldn't know they are different apps. So here's what would happen with automatic updates then: "you have `com.foobar` installed with `versionCode: 5`? Oh, there's a `com.foobar` with `versionCode: 6`, let me update that for you…"

With different tag naming patterns, I can tell my updater which one to pick, so users of my repo wouldn't receive "accidental cross-updates". But I could not serve both apps simultaneously if they share the identical applicationId.

IzzySoft, 2022-07-07:
Well, many people wonder many things. If F-Droid provided those features, they'd wonder the other way around – and complain about degraded security :laughing:

Why a downgrade to a lower `versionCode` is no longer possible (it once was) you must ask those who made that rule. Probably because that *could* break things (imagine an upgrade changed the structures of your local SQLite database. How'd you expect the older version to deal with that?). Plus honestly, how often have you needed to downgrade? The very few times I had, I simply uninstalled and restored from backup.

So I'll stick with the `MkIIIv` branch/flavor then for my repo. How can we ensure I don't pick an APK from the other one accidentally? Please remember that possibility should you one day decide to drop the suffix. And will you consider increasing `versionCode`?

IzzySoft, 2022-07-08:
Thanks! But making the code more complex for a single app is not what I'd go for (especially when it turns out that part does work – but the code update would break it for the 800+ other apps in [the repo](https://apt.izzysoft.de/fdroid) :wink:) No, the only thing missing now is the update notification for the users of the app. Which cannot be fixed unless the `versionCode` will be increased.

IzzySoft, 2022-08-13:
Just picking up the question again, as today my updater once more informed me about there being a problem with `versionCode` and I had to search why (cannot remember details for all the apps, apologies): Will you increase `versionCode` properly with the MKIII `redist` APK at least, which is the one I pick? Those users who want a not-changing `versionCode` could come to your releases section and pick another one manually, as they need to update manually anyway. But those who prefer update notifications and maybe even automated updates could still receive them via my repo then.

IzzySoft, 2023-02-16:
It doesn't make much sense to put an app in a repo if it doesn't offer updates:

* as `versionCode` is never increased, there won't be any update notifications
* even if the user checks manually, there won't be any "update" button (for the same reason); and even though F-Droid offers "downgrades", that possibility doesn't exist either as there are no "previous versions" due to all of them being "the same"
* so all left is to manually download the APK file and side-load it – which is not how an app-repo works or what it is intended for; essentially concerning this specific app it doesn't make much difference if it's in the repo or not, apart for initially discovery and install

So @green-green-avk may I suggest a last time to have an APK with proper versioning? If you prefer it without, maybe just attach it as a separate APK, and mark it with "proper versioning for UPGRADABILITY" :speak_no_evil:

green-green-avk, 2023-02-28:
Sorry for the late response.

Ok. I'll add a new <kbd>fdroid</kbd> flavor for both branches in the next release (in case you prefer <kbd>MkIV-dev</kbd> once).

IzzySoft, 2023-02-28:
Thanks a lot! Please let me know when it's there so I can adjust my updater accordingly to pull the right files.

green-green-avk, 2023-03-08:
Ready.

`*-versioned-*` flavors have the standard versioning.

`*-downgradable-*` flavors uses the `releaseVersionCode` application metadata bundle field.
<https://github.com/green-green-avk/AnotherTerm/blob/f9806e5a5bdb02255b6ab63c3ffe80ad30773fb2/app/src/downgradable/AndroidManifest.xml#L1-L10>

IzzySoft, 2023-03-08:
Thanks a lot! :star_struck: Adjusted the config here and triggered an update check: updater no longer complains, users will receive update notifications – wonderful!

```
$ iod repo get green_green_avk.anotherterm.redist
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/releases'
green_green_avk.anotherterm.redist: checking tag 'MkIV_dev41_release'
green_green_avk.anotherterm.redist: checking tag 'MkIIIv74_release'
green_green_avk.anotherterm.redist: lastRelNo set to '74_release', checking for files
Upstream file date (2023-03-08 04:39) is newer than ours (2021-05-02 02:00).
green_green_avk.anotherterm.redist: returning ['74_release','https://github.com/green-green-avk/AnotherTerm/releases/download/MkIIIv74_release/app-redist-versioned-release.apk']
green_green_avk.anotherterm.redist: MkIIIv73/74_release, https://github.com/green-green-avk/AnotherTerm/releases: https://github.com/green-green-avk/AnotherTerm/releases/download/MkIIIv74_release/app-redist-versioned-release.apk
- Grabbing update for green_green_avk.anotherterm.redist: OK
- Checking 'repo/green_green_avk.anotherterm.redist_133.apk' for libraries and malware …
green_green_avk.anotherterm.redist: check if repo contains FUNDING.yml
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/.github'
green_green_avk.anotherterm.redist: Github reports "Not Found" for https://api.github.com/repos/green-green-avk/AnotherTerm/contents/.github
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/'
green_green_avk.anotherterm.redist: no FUNDING.yml detected.
green_green_avk.anotherterm.redist: calling 'getFastlaneMeta(github,[host:github.com,owner:green-green-avk,repo:AnotherTerm,path:/fastlane/metadata/android])'
green_green_avk.anotherterm.redist: FastlaneFeatures shortdesc,fulldescMD,icon
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/fastlane%2Fmetadata%2Fandroid'
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/fastlane%2Fmetadata%2Fandroid%2Fde'
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/fastlane%2Fmetadata%2Fandroid%2Fen-US'
green_green_avk.anotherterm.redist: looking for 'https://api.github.com/repos/green-green-avk/AnotherTerm/contents/fastlane%2Fmetadata%2Fandroid%2Fen-US%2Fimages'
green_green_avk.anotherterm.redist: checking locale 'de'
green_green_avk.anotherterm.redist: checking locale 'en-US'
```

PS: If you want to add the link next to the PlayStore one: `https://apt.izzysoft.de/packages/green_green_avk.anotherterm.redist/` (you could also [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) for that if you wish).

closed 2023-03-08 by IzzySoft