green-green-avk/AnotherTerm #31: Audio in graphical session

dvlahakis, 2022-06-18:
Are there any suggestions for how audio might work in a proot graphical session? I tried running pulseaudio and pacmd unsuccessfully in a Debian bullseye session. I thought I might be able to stream to the app pulsedroid. Thanks!

green-green-avk, 2022-07-04:
I'll think about it but it is not a priority now...
The main priority is a **PRoot** => **QEMU** migration as Android become more and more restrictive.
Also I recommend you to be mentally ready to a moment when Android own console support will remain in the GitHub releases only (due to tightening of the Google Play rules).

zongou, 2023-10-14:
### pulseaudio tcp & rtp server

- [PulseDroidRtp](https://github.com/wenxin-wang/PulseDroidRtp) (recommanded)
- [SimpleProtocolPlayer](https://github.com/kaytat/SimpleProtocolPlayer) (has sensible latency)

```sh
PORT=4010
RATE=48000
CHANNELS=2
FORMAT=s16le
HOST=127.0.0.1

## Kill previous run
if pulseaudio --check; then
    pulseaudio --kill
fi

## Start pulseaudio service
dbus-launch pulseaudio \
    --verbose \
    --log-level=4 \
    --exit-idle-time=-1 \
    --daemonize=true \
    --load="module-native-protocol-tcp"

## tcp server
pactl load-module module-simple-protocol-tcp \
    rate="${RATE}" \
    format="${FORMAT}" \
    channels="${CHANNELS}" \
    source=auto_null.monitor \
    record=true \
    port="${PORT}" \
    listen="${HOST}"

## rtp server
pactl unload-module module-null-sink
pactl unload-module module-rtp-send

pactl load-module module-null-sink \
    sink_name=rtp \
    format="${FORMAT}" \
    channels="${CHANNELS}" \
    rate="${RATE}"

pactl load-module module-rtp-send \
    source=rtp.monitor \
    destination="${HOST}" \
    port="${PORT}" \
    mtu=320

```

green-green-avk, 2023-12-23:
A related question: <https://github.com/green-green-avk/AnotherTerm/issues/52>

Maybe I will change my mind but rather prefer to implement as [**PipeWire**](https://pipewire.org/).