green-green-avk/AnotherTerm #34: Support styled and colored underlines

oblitum, 2022-12-19:
Hi, I'm not currently an AnotherTerm user, I'm currently using Termux. There's a feature I'm missing on Termux which you might be interested at implementing here, given that I see development is actually more active here than there at the moment.

See: https://github.com/termux/termux-app/issues/870

oblitum, 2022-12-19:
I'm not sure whether AnotherTerm already supports custom fonts, Termux does, and I rely on it for having Nerd Fonts with Powerline glyphs.

green-green-avk, 2022-12-24:
It is from the [**kitty**'s Terminal protocol extensions](https://sw.kovidgoyal.net/kitty/protocol-extensions/).
Why not to add? It will be done.

At the moment, you can also try this trick though:
```sh
echo -e '\e[97mX\e[1;31m\xE1\xAA\xB6\e[0m'
```
It uses the ability to set separate colors of the combining glyphs. (**Another Term** only, it seems...)
The combining glyph here is [<kbd>U+1AB6</kbd>](https://www.compart.com/en/unicode/U+1AB6).

oblitum, 2022-12-25:
Eager to try your impl when it's done.