green-green-avk/AnotherTerm #35: Terminal color is too dark to see

zongou, 2022-12-20:
terminal shows gray for white, and darker blue for blue and makes the status bar of nnn nearly invisible
[image]
[image]

green-green-avk, 2022-12-24:
I took a time for this bug finally posted...

Ok, I'll implement it.

green-green-avk, 2022-12-28:
BTW, too bright regular blue could make eyes bleeding: just try **Midnight Commander** for example.
I wonder, why the author did not make this text bold - all terminals make it bright (not only bold) in this case.
(And, what's more important, why does the `NNN_COLORS` not provide a way to configure it bold?)

Anyways, you could just set `TERM=xterm-256color` and set the color to any 8-bit indexed value [via `NNN_COLORS`](https://manpages.debian.org/testing/nnn/nnn.1.en.html#ENVIRONMENT).
*See the color numbers [here](https://en.wikipedia.org/wiki/ANSI_escape_code#8-bit) (or [here](https://user-images.githubusercontent.com/1482942/93023823-46a6ba80-f5e1-11ea-9ea3-6a3c757704f4.png) as recommended by the author).*

zongou, 2022-12-28:
[image]
[image]
https://github.com/V01dDweller/color-tests
anotherterm does not handle color well

zongou, 2022-12-28:
i have used custom color for nnn after i posted this issue, to make status bar visible

zongou, 2022-12-28:
> BTW, too bright regular blue could make eyes bleeding: just try **Midnight Commander** for example. I wonder, why the author did not make this text bold - all terminals make it bright (not only [...]

[image]

font style is in bold by default

green-green-avk, 2022-12-28:
> anotherterm does not handle color well

A long time ago, I decided to save some memory, so the color attributes are packed in a pretty insane manner now: the upcoming update will fix it.

green-green-avk, 2023-01-10:
Does it look better now?
Please, recheck.

zongou, 2023-01-10:
it is better now, thanks

closed 2023-01-10 by green-green-avk