green-green-avk/AnotherTerm #36: $TERMSH command to operate clipboard

zongou, 2023-01-14:
with this we can sync clipboard between android, will make text editing with editors such as micro and vim much easier.
currently im using $TERMSH send --text--stdin and android app of copy to clipboard to copy selections to android

zongou, 2023-01-14:
https://user-images.githubusercontent.com/23202896/212481334-b71877c8-d2dd-4367-b9f1-6de85af79c0b.mp4

green-green-avk, 2023-01-16:
🤔... Yes. And one more local shell permission for it.

green-green-avk, 2023-01-18:
Ready: <https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_clipboard-copy>

Please, confirm.

zongou, 2023-01-20:
> Ready: https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_clipboard-copy
> 
> Please, confirm.

thanks, it works

closed 2023-01-21 by green-green-avk