green-green-avk/AnotherTerm #39: Charset list is not displayed correctly in terminal

zongou, 2023-02-08:
https://user-images.githubusercontent.com/23202896/217409643-a90e1170-7d07-46b5-abdd-6c54fffdcb82.mp4

AnotherTerm version: MkIV-dev36
device: mi8(dipper)
android version: 10

green-green-avk, 2023-02-08:
I need to find a way how to reproduce it in my environment...
Is it either the original Android&nbsp;10 from the vendor or something custom?

In which version it worked properly?
It could become broken since <kbd>MkIIIv65</kbd> (with the **androidx** library update) but it was a long time ago.

zongou, 2023-02-08:
> I need to find a way how to reproduce it in my environment... Is it either the original Android 10 from the vendor or something custom?
> 
> In which version it worked properly? It could become broken since MkIIIv65 (with the **androidx** library update) but it was a long time ago.

My cellphone runs on MIUI, yes, heavily customized. But I think it's not relevant to that, I have tested it on nearly original Android os( not totally original) with virtual machine software(VMOS is its name), with it i have tested on android 9 and android 7, the view break on android 9 but works fine on android 7

zongou, 2023-02-08:
https://user-images.githubusercontent.com/23202896/217440537-5d755909-6fdf-4e69-89dd-0dec4b0d5b30.mp4

i selected charset GBK cause it seems obvious

green-green-avk, 2023-02-08:
Let's keep some track of reproducibility:

Affected:
* [Xiaomi&nbsp;Mi&nbsp;8](https://www.gsmarena.com/xiaomi_mi_8-9065.php) (Android&nbsp;10 MIUI)

Not affected:
* [Samsung&nbsp;Galaxy&nbsp;Note&nbsp;2 (SCH-I605)](https://www.gsmarena.com/samsung_galaxy_note_ii_cdma-5152.php) (Android&nbsp;4.4.2)
* [Samsung&nbsp;Galaxy&nbsp;Note&nbsp;8 (SM-N950U1)](https://www.gsmarena.com/samsung_galaxy_note8-8505.php) (Android&nbsp;9)
* [Google&nbsp;Pixel&nbsp;7 (GVU6C)](https://www.gsmarena.com/google_pixel_7-11903.php) (Android&nbsp;13)
* Android Studio Emulator image of Android&nbsp;10 x86_64

green-green-avk, 2023-02-14:
## Current implementation details

The character set menu is being opened here:
<https://github.com/green-green-avk/AnotherTerm/blob/c22123dca01c31de1215967d934762eff99710c4/app/src/main/java/green_green_avk/anotherterm/AnsiConsoleActivity.java#L842-L863>

It uses layouts:
* Android&nbsp;4.2 and higher
<https://github.com/green-green-avk/AnotherTerm/blob/c22123dca01c31de1215967d934762eff99710c4/app/src/main/res/layout-v17/dialogmenu_entry.xml#L1-L12>
* All other (derelict) devices
<https://github.com/green-green-avk/AnotherTerm/blob/c22123dca01c31de1215967d934762eff99710c4/app/src/main/res/layout/dialogmenu_entry.xml#L1-L13>

The background is just a stateful color:
<https://github.com/green-green-avk/AnotherTerm/blob/c22123dca01c31de1215967d934762eff99710c4/app/src/main/res/drawable/bg_menu_checkable.xml#L1-L6>

[`CheckedTextView`](https://developer.android.com/reference/android/widget/CheckedTextView) here is automatically substituted by [`AppCompatCheckedTextView`](https://developer.android.com/reference/androidx/appcompat/widget/AppCompatCheckedTextView) during inflation.

zongou, 2023-02-14:
i have noticed the width of text changes. wish this helps

https://user-images.githubusercontent.com/23202896/218643928-02b0dbe5-7a01-4383-825f-2febda851c6a.mp4