green-green-avk/AnotherTerm #42: armeabi-v7a re-support

8itlew7r, 2023-07-18:
I can't install Linux under PRoot, can you please support armeabi-v7a again? Otherwise I won't be able to use your application, which I mainly use to learn programming. If not, thank you for your incredible work!

green-green-avk, 2023-07-18:
At which point `armeabi-v7a` stopped working and how?
*Some symptoms appreciated.*

8itlew7r, 2023-07-19:
I'm sorry, I'm trying to follow your [instructions](https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-under-proot.html#main_content), but all the distributions only support amd64, arm64.

[image]

[image]

green-green-avk, 2023-07-19:
# What happened

About mentioned [instructions](https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-under-proot.html#making-it-quick-linuxcontainersorg-to-the-rescue): I only provided a link to [a third-party](https://linuxcontainers.org/) root-fs builds for their own container system as it was seemingly the best way to do the job at that distant time.
Alas, they have stopped support any 32-bit archs.

# What to do

See <https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-under-proot.html#other-prebuilt-linux-rootfses-list>

Note 1: Where is an automatic install script that you probably want? Nah... Not such quick. I'll try to provide some soon. Use the [**Force**](https://en.wikipedia.org/wiki/The_Force) till this moment and praise the [Omnissiah](https://warhammer40k.fandom.com/wiki/Machine_God).

Note 2: And remember: **Android** prohibits hard links creation by its apps, [**minitar**](https://github.com/green-green-avk/build-libarchive-minitar-android) (the part of the [Android10Essentials plugin](https://github.com/green-green-avk/AnotherTermShellPlugin-Android10Essentials)) just translates all subsequent references to a file content into symlinks to the first encountered reference (that is being written as a file) that usually works but is semantically inaccurate. **PRoot** does it much better. That's why I recommend you to create or obtain only a minimalistic root-fs and then install all the necessary packages.

## Some brief excerpts from this recommendation

### Alpine

Possibly <https://www.alpinelinux.org/downloads/> (see the **"Mini root filesystem"** list) will help you.

### Debian

1) May the **Force** be with you (Alas, a PC is required):
* <https://wiki.debian.org/EmDebian/CrossDebootstrap>
* <https://wiki.ubuntu.com/ARM/RootfsFromScratch/QemuDebootstrap>

2) Just upgrade from some extremely old one:
* <https://github.com/CypherpunkArmory/UserLAnd-Assets-Debian/tree/master/assets>
* <https://download.openvz.org/template/precreated/>

8itlew7r, 2023-07-19:
Sorry, it seems like I'm stuck on this commnand `$("$TERMSH" plugin green_green_avk.anothertermshellplugin_android10essentials minitar) < some.tar.xz`, I don't know how to do the proot part or I really don't know what I'm doing. Can't I just modify the automatic install script to use a custom root-fs zip file?

I can wait for "soon" if that is your suggestion.

8itlew7r, 2023-07-19:
I hardcoded the root-fs zip file into the script and it looks like it worked!

spoelstraethan, 2023-08-08:
postmarketOS is based on Alpine and still has some armv7 aka armhf support I believe.

8itlew7r, 2023-08-11:
> postmarketOS is based on Alpine and still has some armv7 aka armhf support I believe.

I am trying to find the location for rootfs (arm7) but it seems I am having a hard time finding it, do you have a link somehow? By the way, you don't need to bother, I'm already fine with my favorite distro (Alpine and Debian).

spoelstraethan, 2023-08-11:
This might have been related?

https://wiki.postmarketos.org/wiki/Troubleshooting:dependencies

8itlew7r, 2023-08-12:
> This might have been related? https://wiki.postmarketos.org/wiki/Troubleshooting:dependencies
> […](#)

Thank you!