green-green-avk/AnotherTerm #43: Weird error when trying to excecute sh ./repo-linuxcontainers.sh

rektpartyaftermath, 2023-07-20:


green-green-avk, 2023-07-21:
1) Is it one of the latest versions?
* <https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIV_dev47_release>
* <https://github.com/green-green-avk/AnotherTerm/releases/tag/MkIIIv80_release>

2) What are the Android version and manufacturer of your device?

PS:

I could suggest that your device is very old because you allegedly have outdated root certificates:
> `java.security.cert.CertPathValidatorException: Trust anchor for certification path not found.`

See <https://green-green-avk.github.io/AnotherTerm-docs/issues.html#termsh-ssl-handshake-error-on-very-old-androids> if it is true. I recommend you the second solution from the list: <https://android.stackexchange.com/questions/242528/how-to-install-another-term-an-alternative-to-termux-for-android-6-0-and-below/246596#246596>

***Note to myself:** Maybe create and maintain a current bionic `wget` version with an **embedded root CA**.*

PPS:

This linker warning could be safely ignored:
* `#define DF_1_NOW    0x00000001` - all symbol must be resolved at load time - ***Ok***
* `#define DF_1_PIE    0x08000000` - Position Independent Executable (PIE) - ***The cause of this warning on your old 'droid***

***Note to myself:** Maybe I need to compile the `libtermsh.so` without PIE for the `oldgood` flavor.*

green-green-avk, 2023-07-21:
Hmm...

PS and PPS guesses are confirmed then.

Just try the steps described here: <https://android.stackexchange.com/questions/242528/how-to-install-another-term-an-alternative-to-termux-for-android-6-0-and-below/246596#246596>

rektpartyaftermath, 2023-07-22:
Ok then

closed 2023-07-23 by rektpartyaftermath