green-green-avk/AnotherTerm #44: lsusb after libusb setup crashes debian bookworm root session

spoelstraethan, 2023-08-08:
I'm trying to interface with a USB device and Termux wasn't really helping, so when I found your excellent tool I decided to try it out. My first attempt with Alpine had some other issues with the software I was trying to compile but I got things built under Debian, but now when I try to list the devices I get the permission prompt and can accept it, and it prints the correct device info, but then immediately kills the session which must close the USB connection because running the compiled application in a new session reports it can't find a compatible device.

green-green-avk, 2023-08-08:
# What to blame

## What happened

ART / Dalvik (I have no idea of your Android device) VM crash.

## How does it work

The [**libusb** wrapper library](https://github.com/green-green-avk/libusb/tree/v1.0.26-android-libusbmanager) accesses [**LibUsbManager**](https://github.com/green-green-avk/LibUsbManager) to:

1) Receive the device list (and keep connection open to receive plug/unplug notifications).
2) Receive any particular USB device descriptor in order to obtain its meta data / interact with it.
3) When done, the client process just closes the descriptor without any interaction with **LibUsbManager**.

## Where it could crash

Taking into account that you has an output about at least one device:
1) If you have more than one device, **LibUsbManager** could fail with opening subsequent devices.
2) If you have only one device, **LibUsbManager** fail on notification connection close is rather unlikely... 🤔 Did you experience any not USB related terminal crashes?

# What to do

## [`adb`](https://developer.android.com/tools/adb) to the rescue

### For Android 7 and later

Start collecting log:

```sh
adb <device-selection-options> shell 'logcat --pid=$(pidof -s <package-name>) "*:E"'
```

where:

* **\<device-selection-options\>** - just see the `adb` [manual](https://developer.android.com/tools/adb) and `adb --help` (may be omitted if only one device is attached (try `adb devices`)).
* **\<package-name\>** - see your Another Term About screen because different flavors are named differently.

and try to reproduce the issue during it.

### For earlier versions

Please, say me what version you have.

# PS

I'm just curious, what process keeps holding the device descriptor after the VM crash (also Android 7 and later only):

```sh
adb <device-selection-options> shell 'ps -fu $(dumpsys package <package-name> | sed -nre "s/.*userId=([0-9]+)/\\1/p")'
```

green-green-avk, 2023-08-10:
### Another possible cause or "Hello, Android 10"

```
fdsan: attempted to close file descriptor <CENSORED>, expected to be unowned, actually owned by ParcelFileDescriptor <CENSORED>
```

I need one more day to recheck.

spoelstraethan, 2023-08-10:
The crashes were happening on Android 13 on a Pixel 7a, so definitely could be some newer platform changes causing issues. 

`MkIIIv80 green_green_avk.anotherterm` was the first version I tried. I just tried `MkIV-dev47 redist` from GitHub  to see if that makes a difference, but it crashes on `lsusb` or `lsusb -v` as well.

I was able to use `MkIV-dev47 oldgood` on an Android 10 device (Cosmo Communicator) to interact with the same USB device without crashing, but one of the functions of the management software wasn't working, possibly because the USB interface wasn't showing all the details it expected (the lsbusb output in the Debian bookworm PRoot under AnotherTerm is slightly different than on a native Linux system).

I can open a new issue for that so we can dig in a bit more without polluting this crash beyond "it mostly worked on Android 10".

green-green-avk, 2023-08-10:
Yes, please open a new one.

The fix for this one is almost there: <https://github.com/green-green-avk/LibUsbManager/commit/7a1c4f268c934876ebd6146c6885a13e46fafd9a>

green-green-avk, 2023-08-10:
In review on Google Play now...

green-green-avk, 2023-08-12:
Should be available on Google Play now.

spoelstraethan, 2023-08-13:
I tested the version from GitHub and it is working, I'll go check the Play Store one shortly but I'm pretty confident it should work as well as long as the Android 10 plugin is also installed.

green-green-avk, 2023-08-31:
Fixed due to my own tests.

closed 2023-08-31 by green-green-avk