green-green-avk/AnotherTerm #46: Text direction

av10one, 2023-09-17:
Connecting to a VT220 telnet host displaying Hebrew language, the text displayed in the wrong direction - 

[image]

Trying this with another app, it displayed correctly - 

[image]

It'll be great if you could fix it.
To reproduce this I captured this screen to a file - 
[1.txt](https://github.com/green-green-avk/AnotherTerm/files/12643643/1.txt)
Then sent it as HEX to the app.

Thanks

green-green-avk, 2023-09-23:
`konsole`:

[image]

`xterm`:

[image]

🤔 I'll try to support.

av10one, 2023-09-25:
While I don't think I can assist with coding, if you feel I can help in other aspects, see me as fully committed😁.

green-green-avk, 2023-11-13:
Should be fixed now...

The *Raw Mode* is here now (see details in the built-in help):

*Global default option*
[image]

*Per-profile*
[image]

*During session*
[image]

av10one, 2023-11-19:
Is non-monospace character length bug makes it look not alligned like this

[image]

green-green-avk, 2023-11-24:
Yes.

green-green-avk, 2023-12-07:
https://github.com/green-green-avk/AnotherTerm/issues/50#issuecomment-1845607679

closed 2023-12-16 by green-green-avk