green-green-avk/AnotherTerm #52: Audio Support

8itlew7r, 2023-12-21:
Greetings! I'm an IT student utilizing your app to delve into programming in general. In this second term, we're exploring the creation of GUIs in Java using Swing and venturing into crafting a 2D game for finals. Luckily, AnLinux provides helpful scripts for running a Window Manager (via VNC) inside the proot, allowing me to experiment with GUIs. The challenge, however, lies in playing audio/sound. Could you guide me step-by-step on achieving this? Is it possible within the VNC environment? Your assistance would be greatly appreciated.

green-green-avk, 2023-12-22:
This issue looks like a duplicate of <https://github.com/green-green-avk/AnotherTerm/issues/31>.
It has also a possible solution.

BTW, **VNC** is slow, GitHub [<kbd>MkIV-dev</kbd>](https://github.com/green-green-avk/AnotherTerm/releases) releases have some [Wayland](https://green-green-avk.github.io/AnotherTerm-docs/graphical-sessions.html#main_content) support.

8itlew7r, 2023-12-22:
I appreciate the Wayland tip. Would it be appropriate to close this issue now?