green-green-avk/AnotherTerm #53: Shared Memory Error: Can't run Firefox & Falkon

8itlew7r, 2023-12-26:
I'm encountering an issue with shared memory on my non-rooted Android phone running [AnLinux Debian rootfs armhf](https://raw.githubusercontent.com/EXALAB/AnLinux-Resources/master/Rootfs/Debian/armhf/debian-rootfs-armhf.tar.xz).
Despite a potential lapse in installing [shared memory APIs emulation](https://green-green-avk.github.io/AnotherTerm-docs/installing-linux-apis-emulation-for-nonrooted-android.html#main_content),  Xwayland seems to be functioning.

I attempted an upgrade to bookworm, but the problem persists.
Any insights or guidance would be greatly appreciated.

[image]

```
my_acct@localhost:~# echo $LIBWRAPDROID_SOCKET_NAME
green_green_avk.anotherterm.oldgood.reswrap.41ab53acf87b4834b520769ff6ed4e45

my_acct@localhost:~# echo $LIBWRAPDROID_AUTH_KEY
b3625515eeec4055b22a3a8c9fcfa077

my_acct@localhost:~# echo $LD_PRELOAD
/opt/shm/lib/libwrapdroid-shm-sysv.so:/opt/shm/lib/libwrapdroid-shm-posix.so

my_acct@localhost:~# falkon
falkon: error while loading shared libraries: libx264.so.160: cannot enable executable stack as shared object requires: Permission denied

my_acct@localhost:~# firefox-esr
[Parent 8028, Main Thread] WARNING: read-only dup failed (Permission denied); not using memfd: file ./ipc/chromium/src/base/shared_memory_posix.cc:255
[Parent 8028, Main Thread] WARNING: failed to open shm: Permission denied: file ./ipc/chromium/src/base/shared_memory_posix.cc:361
[Parent 8028, Main Thread] WARNING: failed to open shm: Permission denied: file ./ipc/chromium/src/base/shared_memory_posix.cc:361
ExceptionHandler::GenerateDump cloned child 8071
ExceptionHandler::WaitForContinueSignal waiting for continue signal...
ExceptionHandler::SendContinueSignalToChild sent continue signal to child

my_acct@localhost:~# neofetch
       _,met$$$$$gg.          my_acct@localhost
    ,g$$$$$$$$$$$$$$$P.       --------------
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 11 (bullseye) armv8l
 ,$$P'              `$$$.     Kernel: 4.9.190-perf-gd642c72
',$$P       ,ggs.     `$$b:   Uptime: 19716 days, 9 hours, 16 mins
`d$$'     ,$P"'   .    $$$    Packages: 892 (dpkg)
 $$P      d$'     ,    $$P    Shell: bash 5.1.4
 $$:      $$.   -    ,d$$'    Terminal: libproot.so
 $$;      Y$b._   _,d$P'      CPU: MT6762G (8) @ 2.001GHz
 Y$$.    `.`"Y$$$$P"'         Memory: 1029MiB / 1777MiB
 `$$b      "-.__
  `Y$$
   `Y$$.
     `$$b.
       `Y$$b.
          `"Y$b._
              `"""

my_acct@localhost:~# apt search falkon | grep "falkon"
falkon/oldstable,now 3.1.0+dfsg1-11 armhf [installed]

my_acct@localhost:~# apt search firefox-esr | grep "firefox-esr"
firefox-esr/oldstable-security,now 115.6.0esr-1~deb11u1 armhf [installed]

my_acct@localhost:~# echo $APP_VERSION
MkIV-dev52

my_acct@localhost:~# echo $MY_DEVICE_ABIS
armeabi-v7a armeabi

my_acct@localhost:~# echo $MY_ANDROID_SDK
29

```

On the session running Xwayland (I have a hard time describing the screenshot below), 
there is an error that seems to occur only specifically when trying to start `firefox-esr`.
In addition, the Falkon release on Debian bookworm presents a bug related to `/dev/shm` permissions, 
with a suggested solution of changing the permission of this path to 1777.

[image]

```
Helper: Starting...
=! Can't get peer creds: Invalid argument
=! Can't get peer creds: Invalid argument
=! Can't get peer creds: Invalid argument
=! Can't get peer creds: Invalid argument
=! Can't get peer creds: Invalid argument
=! Can't get peer creds: Invalid argument

```

I'm not sure if this is relevant, but I got an stderr message while running `( S=install-libwrapdroid.sh ; "$TERMSH" copy -f -fu "https://raw.githubusercontent.com/green-green-avk/AnotherTerm-scripts/master/$S" -tp . && chmod 755 $S && ./$S )`
```
debconf: delaying package configuration, since apt-utils is not installed
Cloning into 'libwrapdroid'...
server.c: In function 'main':
server.c:396:20: warning: unused parameter 'argc' [-Wunused-parameter]
  396 | int main(const int argc, char **const argv) {
      |          ~~~~~~~~~~^~~~
server.c:396:39: warning: unused parameter 'argv' [-Wunused-parameter]
  396 | int main(const int argc, char **const argv) {
      |                          ~~~~~~~~~~~~~^~~~
At top level:
server.c:69:12: warning: 'readString' defined but not used [-Wunused-function]
   69 | static int readString(const int sock, char *const buf, const uint16_t len) {
      |            ^~~~~~~~~~
server.c:65:12: warning: 'writeField' defined but not used [-Wunused-function]
   65 | static int writeField(const int sock, const void *const v, const size_t l) {
      |            ^~~~~~~~~~
common.c: In function 'sendFds':
common.c:61:10: warning: variable-length array bound is unknown [-Wvla-larger-than=]
   61 |     char cmsg_buf[cmsg_space] __attribute__((aligned(_Alignof(struct cmsghdr))));
      |          ^~~~~~~~
In file included from shm-sysv.c:7:
common-cli.h:22:12: warning: 'writeString' defined but not used [-Wunused-function]
   22 | static int writeString(const int sock, const char *const v) {
      |            ^~~~~~~~~~~
common-cli.h:18:12: warning: 'writeU32' defined but not used [-Wunused-function]
   18 | static int writeU32(const int sock, const uint32_t v) {
      |            ^~~~~~~~

```

green-green-avk, 2024-01-05:
# Firefox

Ah, yes:

<https://github.com/termux/proot/issues/162>

```bash
export MOZ_FAKE_NO_SANDBOX=1
```

8itlew7r, 2024-01-07:
The problem persists, and I have a hunch that it could be connected to the shared memory API.

green-green-avk, 2024-01-22:
The problem persists in the same way or something have changed?

8itlew7r, 2024-01-25:
> The problem persists in the same way or something have changed?

I'm sorry, but the problem persists in the same way. I was wondering if, by any chance, it might have worked on your end, maybe with some emulation of my device properties?