green-green-avk/AnotherTerm #56: openssh-server  install error in ubuntu jammy

myan-o, 2024-02-09:
```
Setting up openssh-server (1:8.9p1-3ubuntu0.6) ...##....]
Progress: [ 93%] [###################################...]
Creating config file /etc/ssh/sshd_config with new version
Creating SSH2 RSA key; this may take some time ...
3072 SHA256:Y5wqaMdBos+U5AJXPuREqH/nKGis0V7WjIbD1ZfMxIA root@localhost (RSA)
Creating SSH2 ECDSA key; this may take some time ...
256 SHA256:RK4PKkcPVHV9t9C+u+7nr9kXJUO6l26LJ/O1/nlSsqE root@localhost (ECDSA)
Creating SSH2 ED25519 key; this may take some time ...
256 SHA256:DDMImFvTmzF61ebHyG6RyyMTYVrO68GAjJzjO07zjRE root@localhost (ED25519)
useradd: /etc/passwd.29268: lock file already used
useradd: cannot lock /etc/passwd; try again later.
adduser: `/sbin/useradd -d /run/sshd -g nogroup -s /usr/sbin/nologin -u 106 sshd' returned error code 1. Exiting.
dpkg: error processing package openssh-server (--configure):
 installed openssh-server package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
Errors were encountered while processing:
 openssh-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@localhost:~#
```

myan-o, 2024-02-09:
I was able to start it after adding it manually.
```
echo "sshd:x:33:33:sshd:/var/empty:/usr/sbin/nologin" >> /etc/passwd
service ssh start
```

closed 2024-02-10 by myan-o

green-green-avk, 2024-02-26:
BTW, I saw it once with a **Pixel 7 Pro** but never with **Pixel 7**. I'll investigate when I'll have some time and a device to reproduce it.