green-green-avk/AnotherTerm #57: pip install error in ubuntu jummy

myan-o, 2024-02-09:
```
root@localhost:~/git# pip3 install pio
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_vendor/platformdirs/android.py", line 85, in _android_folder
    from jnius import autoclass
ModuleNotFoundError: No module named 'jnius'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/main.py", line 9, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/main_parser.py", line 8, in <module>
    from pip._internal.cli import cmdoptions
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/cmdoptions.py", line 23, in <module>
    from pip._internal.cli.parser import ConfigOptionParser
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/parser.py", line 12, in <module>
    from pip._internal.configuration import Configuration, ConfigurationError
  File "/usr/lib/python3/dist-packages/pip/_internal/configuration.py", line 26, in <module>
    from pip._internal.utils.logging import getLogger
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/logging.py", line 27, in <module>
    from pip._internal.utils.misc import ensure_dir
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/misc.py", line 39, in <module>
    from pip._internal.locations import get_major_minor_version
  File "/usr/lib/python3/dist-packages/pip/_internal/locations/__init__.py", line 14, in <module>
    from . import _distutils, _sysconfig
  File "/usr/lib/python3/dist-packages/pip/_internal/locations/_distutils.py", line 19, in <module>
    from .base import get_major_minor_version
  File "/usr/lib/python3/dist-packages/pip/_internal/locations/base.py", line 12, in <module>
    USER_CACHE_DIR = appdirs.user_cache_dir("pip")
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/appdirs.py", line 17, in user_cache_dir
    return _appdirs.user_cache_dir(appname, appauthor=False)
  File "/usr/lib/python3/dist-packages/pip/_vendor/platformdirs/__init__.py", line 114, in user_cache_dir
    return PlatformDirs(appname=appname, appauthor=appauthor, version=version, opinion=opinion).user_cache_dir
  File "/usr/lib/python3/dist-packages/pip/_vendor/platformdirs/android.py", line 43, in user_cache_dir
    return self._append_app_name_and_version(_android_folder(), "cache")
  File "/usr/lib/python3/dist-packages/pip/_vendor/platformdirs/android.py", line 97, in _android_folder
    raise OSError("Cannot find path to android app folder")
OSError: Cannot find path to android app folder
root@localhost:~/git#
```

green-green-avk, 2024-02-12:
Uh, huh...

## The problem

`/usr/lib/python3/dist-packages/pip/_vendor/platformdirs/__init.py__`:

In Debian:
```python3
def _set_platform_dir_class() -> type[PlatformDirsABC]:
    if sys.platform == "win32":
        from pip._vendor.platformdirs.windows import Windows as Result
    elif sys.platform == "darwin":
        from pip._vendor.platformdirs.macos import MacOS as Result
    else:
        from pip._vendor.platformdirs.unix import Unix as Result

    if os.getenv("ANDROID_DATA") == "/data" and os.getenv("ANDROID_ROOT") == "/system":

        if os.getenv("SHELL") or os.getenv("PREFIX"):
            return Result

        from pip._vendor.platformdirs.android import _android_folder

        if _android_folder() is not None:
            from pip._vendor.platformdirs.android import Android

            return Android  # return to avoid redefinition of result

    return Result
```

In Ubuntu:
```python3
def _set_platform_dir_class() -> type[PlatformDirsABC]:
    if os.getenv("ANDROID_DATA") == "/data" and os.getenv("ANDROID_ROOT") == "/system":
        module, name = "pip._vendor.platformdirs.android", "Android"
    elif sys.platform == "win32":
        module, name = "pip._vendor.platformdirs.windows", "Windows"
    elif sys.platform == "darwin":
        module, name = "pip._vendor.platformdirs.macos", "MacOS"
    else:
        module, name = "pip._vendor.platformdirs.unix", "Unix"
    result: type[PlatformDirsABC] = getattr(importlib.import_module(module), name)
    return result
```

## The solution

Just unset any of the `ANDROID_DATA` or `ANDROID_ROOT` environment variables and proceed.

BDhackers009, 2024-03-29:
You can use this script...to fix the issue

https://github.com/BDhackers009/fixpip3

green-green-avk, 2024-03-31:
Unsetting `ANDROID_DATA` or `ANDROID_ROOT` environment variables is way more convenient. Mentioned type of vandalizing things is not necessary here at all.