green-green-avk/AnotherTerm #61: Way to use the libusb

BDhackers009, 2024-03-29:
Hello sir, so I just installed anotherterm on my phone. previously I was using Termux. But I wanted to use my phone to flash or program my mcu's like esp32/esp866. I installed libusb following the wiki. But now how should I proceed? I mean how can i get a port to use in tools like esptool. Sorry I'm new to this libusb thing. Any kind of information will help sir. Btw thanks for your effort

green-green-avk, 2024-03-30:
Hmm... So, you need a serial port not the `usbcore` functionality...
Well, I definitely need to add some serial driver emulation into **PRoot** for good...

However as long as `esptool` is a Python script and it uses the `serial` package:

In your project directory:
```sh
mkdir -p 'serial_wrapper'
```

`serial_wrapper/protocol_termsh.py`:
```python3
from __future__ import absolute_import

#import logging
import numbers

from serial.serialutil import SerialBase, SerialException, to_bytes, iterbytes, SerialTimeoutException, PortNotOpenError

import subprocess
import select, fcntl, termios, sys

class Serial(SerialBase):
    """Serial port implementation that simulates a loop back connection in plain software."""

    BAUDRATES = (50, 75, 110, 134, 150, 200, 300, 600, 1200, 1800, 2400, 4800,
                 9600, 19200, 38400, 57600, 115200)

    def __init__(self, *args, **kwargs):
        self.proc = None
        self._cancel_write = False
        super(Serial, self).__init__(*args, **kwargs)

    def open(self):
        """\
        Open port with current settings. This may throw a SerialException
        if the port cannot be opened.
        """
        if self.is_open:
            raise SerialException("Port is already open.")

        # not that there is anything to open, but the function applies the
        # options found in the URL
        self.from_url(self.port)

        # not that there anything to configure...
        self._reconfigure_port()
        # all things set up get, now a clean start
        self.is_open = True
        if not self._dsrdtr:
            self._update_dtr_state()
        if not self._rtscts:
            self._update_rts_state()
        self.reset_input_buffer()
        self.reset_output_buffer()

    def close(self):
        if self.is_open:
            self.is_open = False
            proc = self.proc
            if proc is not None:
                proc.terminate()

    def _reconfigure_port(self):
        """\
        Set communication parameters on opened port. For the loop://
        protocol all settings are ignored!
        """
        # not that's it of any real use, but it helps in the unit tests
        if not isinstance(self._baudrate, numbers.Integral) or not 0 < self._baudrate < 2 ** 32:
[... 132 lines cut ...]

    @property
    def ri(self):
        """Read terminal status line: Ring Indicator"""
        if not self.is_open:
            raise PortNotOpenError()
        return False

    @property
    def cd(self):
        """Read terminal status line: Carrier Detect"""
        if not self.is_open:
            raise PortNotOpenError()
        return True
```

Augment `esptool` with (see <https://github.com/espressif/esptool/blob/ef8ee8aedff5a642c0cf8e10997ad1bb373bf0a2/esptool/loader.py#L30>):
```python3
serial.protocol_handler_packages.append('serial_wrapper')
```

Now (the only) serial port (USB or Bluetooth) on your system could be referred as `termsh://`.

Feel free to improve the `serial_wrapper/protocol_termsh.py` implementation as you wish.
See `termsh serial` info [here](https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_serial).

BDhackers009, 2024-03-30:
First of all thanks a lot sir you came up with a solution that fast. Secondly I'm not good at programming nor very experienced in *nix.

I did create serial_wrapper/protocol_termsh.py inside esptool/esptool/ directory where the loader.py is and then added

``` serial.protocol_handler_packages.append('serial_wrapper')

```

under the import Serial line in line 30. then tried running

``` esptool.py --chip esp32 --port termsh:///dev/bus/usb/001/002 chip_id ```

but it threw some errors. Now I wanna know actually how do I use this cool script ☹️. Could you please help me more sir if you got time. I would be eternally grateful.

green-green-avk, 2024-03-31:
What errors?

BDhackers009, 2024-03-31:
Here it is...

[image]

green-green-avk, 2024-04-01:
Uh, huh... `/dev/tty*` for serial ports emulation by **PRoot** seems desperately required here...

BDhackers009, 2024-04-01:
> Uh, huh... `/dev/tty*` for serial ports emulation by **PRoot** seems desperately required here...

So this is a dead end?

green-green-avk, 2024-04-01:
Hmm...

It seems, the `serial` Python library was made in [a sort of stupid way](https://github.com/pyserial/pyserial/blob/7aeea35429d15f3eefed10bbb659674638903e3a/serial/serialposix.py#L725): there is no API to return the whole modem state in an atomic manner...

Of course, it's possible to change [`UnixTightReset`](https://github.com/espressif/esptool/blob/ef8ee8aedff5a642c0cf8e10997ad1bb373bf0a2/esptool/reset.py#L94) to [`ClassicReset`](https://github.com/espressif/esptool/blob/ef8ee8aedff5a642c0cf8e10997ad1bb373bf0a2/esptool/reset.py#L79) in the **esptool** for this particular case (or make a custom `serial` library with the extended API?) and implement a modem control API by an additional couple of pipes in `termsh serial`... What type of chip your device has? I need to check if this support is already implemented in the [**USBSerial**](https://github.com/felHR85/UsbSerial).

BDhackers009, 2024-04-01:
Its showing CH340 as the COM port. And its an [ESP32](https://www.amazon.com/dp/B0CLR7MQ91). However I do have some other esp32 and esp8266 modules lying around.

[image]

BDhackers009, 2024-05-04:
Sir any update?

9hm2, 2024-06-14:
What type of USB serial chip would you use for device programming?

BDhackers009, 2024-06-15:
> What type of USB serial chip would you use for device programming?
> 
> Try this: https://github.com/9hm2/AnotherTerm_SerialDriver_ttyLike

I saw your demo video. It looks promising. I will try it and let you know. Thanks for the script btw