green-green-avk/AnotherTerm #63: Question on permissions

IzzySoft, 2024-04-29:
For the latest release, my scanner just reported:

```
! repo/green_green_avk.anotherterm.redist_145.apk declares flag(s): usesCleartextTraffic
! repo/green_green_avk.anotherterm.redist_145.apk declares sensitive permission(s):
  android.permission.READ_EXTERNAL_STORAGE android.permission.CAMERA
! repo/green_green_avk.anotherterm.redist_145.apk contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)
```

Could you please clarify their usage? Thanks in advance! (and if you want the other permissions accompanied by an explanation as well, I'll welcome those too :wink:)

Ah, that `DEPENDENCY_INFO_BLOCK` is easily solved by the way:

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains.

green-green-avk, 2024-04-30:
# Short answer first

> `usesCleartextTraffic` & `<certificates src="user" />`

There are

* [`termsh pick`](https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_pick)
* [`termsh copy`](https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_copy)
* [`termsh cat`](https://green-green-avk.github.io/AnotherTerm-docs/local-shell-utility.html#cmd_cat)

that could get content from network. How else the user could access HTTP and HTTPS if the device certificates are outdated?

> `android.permission.READ_EXTERNAL_STORAGE`

It is an ancient one... <https://github.com/green-green-avk/AnotherTerm/issues/1>
There are people for whom SAF (via `termsh`) is not an option...

> `android.permission.CAMERA`

Oh huh: <https://green-green-avk.github.io/AnotherTerm-docs/issues.html#android9-uvc>

> `contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)`

Is it bad??? 🤔

IzzySoft, 2024-04-30:
So let me rephrase, while keeping it simple:

* `usesCleartextTraffic` is used for connections in the local network
* `READ_EXTERNAL_STORAGE` is of course very dang… err, skip that, we're talking about a terminal app here. Sorry, I just quoted the log.
* `CAMERA` is required to connect to any USB_CLASS_VIDEO device since Android 9

I've put that in for now, so these are thereby on the "green list" for your app.

> Is it bad???

Can you tell? Just look what's inside. Wait, you can't? How's that? Oh, only Google can as it's encrypted with their key? Well, [they'd never](
https://techcrunch.com/2022/11/14/google-pay-391-5-million-location-tracking-settlement/ "2022") collect [location data](https://www.latimes.com/california/story/2023-09-14/google-to-pay-93-million-after-state-investigation-finds-company-used-location-data-without-consent "2023") without [your consent](https://beebom.com/google-tracks-location/ "2019"), so they'd certainly not put something strange in here if they say they don't, would they? Apologies for the sarkasm, but: it's a "proprietary blob", and in the FOSS world such are questionable at best. So I'd suggest you keep this out of the APKs here – and if you want, keep it in for the AABs shipped to play. Would that work for you?