green-green-avk/AnotherTerm #67: Black on white/light text rendering missing

gh-ix, 2024-06-04:
Hello,
again, thanks a lot for this great Android app.
Unfortunately I never had good eyes and always found it counter productive working with white-on-black text.
The smaller the fonts are, the harder to read compared to black-on white.

This especially affects devices where AnotherTerm typically runs.
Unfortunately currently there's only a WHITE-on-**black** text mode, which historically only was used because of the way cathode ray tubes are working - the ray needs to energize the glowing pixels, so it was technically straight forward that the light-on-dark is the default.
But that never was ergonomically well default.
Sun made the 'reverse' default even at CRT times :-)  (On LCDs, white-on-black is technically the 'reverse' mode, while OLED again correspond to the CRT mode - and only on OLEDs, there's a real justification  for this mode imho)!

I'd very much appreciate if you could implement a **black**-on-WHITE text rendering mode.
Preferably with adjustable white tint-color, but that's fine tuning only - the white-on-black is a major hindrance for me currently - despite the interestingly well working fine-striped backgrounds (very surprising that background, alleviates the white-on-block mode)

Thanks!

green-green-avk, 2024-06-16:
Hmm... I just trying to imagine how could we  represent some colorful text on some background...

What about a color profile checkbox that makes each color channel $C_c \mid c\in\set{R,G,B}, C_c\in[0,1]$:
* Bright (for background): $C_{c\ \text{bright}}=C_c+\min\limits_{i\in\set{R,G,B}}(1-C_i)$
* Dark (for foreground): $C_{c\ \text{dark}}=C_c-\min\limits_{i\in\set{R,G,B}}(C_i)$

Please, confirm.

gh-ix, 2024-06-17:
Hey, thanks for your attention!

If I correctly understand, you are trying to solve the problem that the color 'palette' doesn't match well for terminal emulators rendering output of colorizing TUI applications on white/light background.
I'm not sure if that's the job for the terminal emulator - thought many times about it, still unsure...

I always avoided adjusting that in the terminal emulatior and kept all non bg/fg colors ANSI defaults.
(Still pending is my test to alter ANSI output at the 'server' side - I wrote a x-shell config framework which takes care about terminfo(5) and composes a uniform and correctly formatted $prompt for each (bourne/korn-like) shell e.g. - incorporating 'themes' at login-shell level seems a valid approch to me - but nobody else does/did as far as I'm aware... and I guess it will interfere with ncusres and the like...)

TUI applications often provide 'themes', and most times offer a 'light' theme, for which colorized text works well on light background.  Those themes usually leave out fg/bg color, so it's a general assumption that the terminal emulator doesn't use divergent color settings.

If your formula results in well readable colorized text (I can't really imagine the look of these 'dark-composed' colors - I guess they are not very colorful in the end), it's definetly a very good option to offer via checkbox, but please make it an **additional option**, so that one can use white/light background without touching the rest of the palette.

Thanks a lot!